PiSupply/PiJuice #543: frequent COMMUNICATION_ERROR issues w/python timelapse camera script

f2calv, 2020-09-12:
Sorry in advance for a lengthy issue post, after days of researching all the issues here (often with lack of critical detail) I figured I'd include _all_ the relevant information in my first post...

I have recently purchased a PiJuice for a timelapse camera project, the Pi will be set-up for headless off-grid/offline operation. During local headless testing I am using a USB wifi dongle, on 'location' there will be generally no connection except an occasional wifi connection from a mobile phone hotspot.

The PiJuice will wake-up the Pi at sunrise, and shut down at sunset - during the day the alarm should Wake-Up every X minutes. I'm using a cronjob on @reboot, so effectively the Pi is up for ~30 seconds then shut down again.

Very often the `pijuice.rtcAlarm.XXXXX` methods fail with COMMUNICATION_ERROR.

I have implemented a retry mechanism similar to #212 however I am not happy with the retry because something clearly isn't working correctly and after trying every combination I need help...

Test hardware/setup;
- Pi A+ or Pi 2B (issue occurs with either Pi)
- PiJuice w/1820mAh supplied battery (firmware 1.4)
- Raspbian Lite (August 2020)
- Pi camera & WiFi dongle
- No other HATs in operation

pijuice_cli;
- have set the RTC clock to the PI datetime initially
- set correct battery profile
 
Raspian;
- PiJuice sw package + supporting python pip3 packages (logging, etc)
- timezone on UTC
- added `dtoverlay=i2c-rtc,ds1307` to the bottom of the config.txt
- changed id-eeprom from 0x50 to 0x52
- removed fake-hwclock `sudo apt-get remove fake-hwclock -y`
- to start the script (with logging) on each boot up I have added `@reboot /usr/bin/python3 /home/pi/github.py >> /home/pi/github.py.log 2>&1` to the crontab

> pi@raspberrypi:~ $ lsmod|grep i2c
> regmap_i2c             16384  1 rtc_ds1307
> i2c_bcm2835            16384  0
> i2c_dev                16384  0

> pi@raspberrypi:~ $ i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --

Below is my slimmed down Python 3 script (github.py) with the critical functions.

Note: I set `rebootLoopEnabled=True` when I want to start testing the reboot loop. When I want to stop testing the reboot loop I upload a new script file with `rebootLoopEnabled=False` during the 20-second 'window' before the next reboot...

```
from pijuice import PiJuice
import os
import sys
import time
from datetime import datetime
import subprocess

rebootLoopEnabled = False
secondsSleep = 20

def init():
    print()  # blank line for logging separation
    print("Checking for /dev/i2c-1", end='')
    attempt = 1
    while not os.path.exists('/dev/i2c-1'):
        time.sleep(0.1)
        attempt += 1
    print(
        "... found /dev/i2c-1 in {attempt} attempt(s)".format(attempt=attempt))

[... 185 lines cut ...]
> 2020-09-12 05:31:19 script completed in 20.44 second(s)
> sudo shutdown -h now <--DISABLED AS TESTING
> pi@raspberrypi:~/timelapse $ python3 github.py
> 
> Checking for /dev/i2c-1... found /dev/i2c-1 in 1 attempt(s)
> Connecting to PiJuice... connection established :)
> Synchronize Pi hardware clock... CompletedProcess(args=['sudo', 'hwclock', '--hctosys'], returncode=0, stdout=b'', stderr=b'')
> GetChargeLevel();
> {'data': 54, 'error': 'NO_ERROR'}
> 2020-09-12 05:31:24 taking photo and doing stuff here (sleeping for 20 seconds)
> rebootLoopEnabled=False
> ClearAlarmFlag()
> {'error': 'COMMUNICATION_ERROR'}
> SetWakeupEnabled(False)
> {'error': 'COMMUNICATION_ERROR'}
> 2020-09-12 05:31:45 script completed in 20.44 second(s)
> sudo shutdown -h now <--DISABLED AS TESTING
> pi@raspberrypi:~/timelapse $
> 

I have researched many of the open/closed issues here from the past years, there seems to be a myriad of conflicting information, maybe someone can tell me what I have missed?

I have a bunch of other questions but they can wait... the COMMUNICATION_ERROR issue is a blocker for my progress on this project right now.

Thanks in advance!

tvoverbeek, 2020-09-12:
@f2calv Thanks for your extensive report.

I tried to reproduce your problem.
I took my Pi 2B updated Raspbian (now Raspberry Pi Os) to the latest version.
Mounted my PiJuice HAT with BP7X on the 2B and installed pijuice-base.
A difference is that I have the desktop version.
In raspi-config I changed to auto-login on console (=> No desktop).
Removed fake-hwclock and also disabled the systemd-timesyncd service.
This to be able to reach the system via WiFi preventing timesyncd messing around with the RTC. 

Repeated runs of your test script with rebootLoopEnabled = False did not produce any COMMUNICATION ERROR

Also did a run with rebootLoopEnabled = True, script started by crontab.
This nicely ran every 2 minutes.
Here an excerpt from my log:
```
Checking for /dev/i2c-1... found /dev/i2c-1 in 1 attempt(s)
Connecting to PiJuice... connection established :)
Synchronize Pi hardware clock... CompletedProcess(args=['sudo', 'hwclock', '--hctosys'], returncode=0, stdout=b'', stderr=b'')
GetChargeLevel();
{'data': 35, 'error': 'NO_ERROR'}
2020-09-12 17:20:00 taking photo and doing stuff here (sleeping for 20 seconds)
rebootLoopEnabled=True
SetAlarm()
Setting PiJuice Wake-Up alarm...
attempt 1 - alarm set for {'year': 'EVERY_YEAR', 'month': 'EVERY_MONTH', 'day': 'EVERY_DAY', 'hour': 'EVERY_HOUR', 'minute': 22, 'second': 0}
SetWakeupEnabled(True)
SetSystemPowerSwitch(0)
2020-09-12 19:20:20 script completed in 20.64 second(s)

Checking for /dev/i2c-1... found /dev/i2c-1 in 1 attempt(s)
Connecting to PiJuice... connection established :)
Synchronize Pi hardware clock... CompletedProcess(args=['sudo', 'hwclock', '--hctosys'], returncode=0, stdout=b'', stderr=b'')
GetChargeLevel();
{'data': 37, 'error': 'NO_ERROR'}
2020-09-12 17:22:18 taking photo and doing stuff here (sleeping for 20 seconds)
rebootLoopEnabled=True
SetAlarm()
Setting PiJuice Wake-Up alarm...
attempt 1 - alarm set for {'year': 'EVERY_YEAR', 'month': 'EVERY_MONTH', 'day': 'EVERY_DAY', 'hour': 'EVERY_HOUR', 'minute': 24, 'second': 0}
SetWakeupEnabled(True)
SetSystemPowerSwitch(0)
2020-09-12 19:22:38 script completed in 20.63 second(s)

Checking for /dev/i2c-1... found /dev/i2c-1 in 1 attempt(s)
Connecting to PiJuice... connection established :)
Synchronize Pi hardware clock... CompletedProcess(args=['sudo', 'hwclock', '--hctosys'], returncode=0, stdout=b'', stderr=b'')
GetChargeLevel();
{'data': 39, 'error': 'NO_ERROR'}
2020-09-12 17:24:18 taking photo and doing stuff here (sleeping for 20 seconds)
rebootLoopEnabled=True
SetAlarm()
Setting PiJuice Wake-Up alarm...
attempt 1 - alarm set for {'year': 'EVERY_YEAR', 'month': 'EVERY_MONTH', 'day': 'EVERY_DAY', 'hour': 'EVERY_HOUR', 'minute': 26, 'second': 0}
SetWakeupEnabled(True)
SetSystemPowerSwitch(0)
2020-09-12 19:24:38 script completed in 20.63 second(s)

Checking for /dev/i2c-1... found /dev/i2c-1 in 1 attempt(s)
Connecting to PiJuice... connection established :)
Synchronize Pi hardware clock... CompletedProcess(args=['sudo', 'hwclock', '--hctosys'], returncode=0, stdout=b'', stderr=b'')
GetChargeLevel();
{'data': 40, 'error': 'NO_ERROR'}
2020-09-12 17:26:19 taking photo and doing stuff here (sleeping for 20 seconds)
rebootLoopEnabled=True
SetAlarm()
Setting PiJuice Wake-Up alarm...
attempt 1 - alarm set for {'year': 'EVERY_YEAR', 'month': 'EVERY_MONTH', 'day': 'EVERY_DAY', 'hour': 'EVERY_HOUR', 'minute': 28, 'second': 0}
SetWakeupEnabled(True)
SetSystemPowerSwitch(0)
2020-09-12 19:26:39 script completed in 20.63 second(s
```
As you can see I was powering the PiJuice during this (chargelevel is increasing).
The time in the script completed messages is local time (I am in Austria, UTC+2).

What is strange is that the non-RTC calls do not give COMMUNICATION ERROR.
Are there any other programs running on your Pi which could access the RTC (timesyncd, ntp, ...)?
What are your wakeup alarm settings in pijuice_cli? Should not be used since you program the alarm yourself.
Anything in System Task? 
I assume you have  Software Halt Power Off enabled with a sufficient delay period.
This ensures the PiJuice will remove the 5V to the Pi after shutdown during the 2 minute period
(extending battery life when running on battery).

f2calv, 2020-09-13:
Hi @tvoverbeek thanks for your efforts.

To answer your questions;
- timesyncd/ntp services - yes were running, I have addressed them (see below).
- in pijuice_cli I only set the battery profile and changed 50->52, I have never touched the wake-up/alarm settings UI
- For Software Halt Power Off I use `pijuice.power.SetPowerOff(20)`
- No additional services should be running outside of what comes with a fresh install of Raspian `ps -A`;

> pi@raspberrypi:~/timelapse/logs $ ps -A
>   PID TTY          TIME CMD
>     1 ?        00:00:09 systemd
>     2 ?        00:00:00 kthreadd
>     3 ?        00:00:00 kworker/0:0-mm_percpu_wq
>     4 ?        00:00:00 kworker/0:0H-kblockd
>     5 ?        00:00:00 kworker/u2:0-phy0
>     6 ?        00:00:00 mm_percpu_wq
>     7 ?        00:00:01 ksoftirqd/0
>     8 ?        00:00:00 kdevtmpfs
>     9 ?        00:00:00 netns
>    10 ?        00:00:00 kworker/0:1-events_power_efficient
>    11 ?        00:00:00 khungtaskd
>    12 ?        00:00:00 oom_reaper
>    13 ?        00:00:00 writeback
>    14 ?        00:00:00 kcompactd0
>    28 ?        00:00:00 kblockd
>    29 ?        00:00:00 blkcg_punt_bio
>    30 ?        00:00:00 watchdogd
>    31 ?        00:00:00 rpciod
>    32 ?        00:00:00 kworker/u3:0
>    33 ?        00:00:00 xprtiod
>    34 ?        00:00:00 kswapd0
>    35 ?        00:00:00 nfsiod
>    36 ?        00:00:00 iscsi_eh
>    37 ?        00:00:00 dwc_otg
>    38 ?        00:00:00 DWC Notificatio
>    39 ?        00:00:01 kworker/0:2-events_freezable
>    40 ?        00:00:00 vchiq-slot/0
>    41 ?        00:00:00 vchiq-recy/0
>    42 ?        00:00:00 vchiq-sync/0
>    43 ?        00:00:00 vchiq-keep/0
>    44 ?        00:00:00 SMIO
>    45 ?        00:00:00 kworker/0:3-events
>    46 ?        00:00:00 mmc_complete
>    47 ?        00:00:01 kworker/0:1H-mmc_complete
>    48 ?        00:00:00 jbd2/mmcblk0p2-
>    49 ?        00:00:00 ext4-rsv-conver
>    50 ?        00:00:00 kworker/0:2H-mmc_complete
>    51 ?        00:00:00 ipv6_addrconf
>    52 ?        00:00:00 kworker/u2:1-events_unbound
>    84 ?        00:00:01 systemd-journal
>    89 ?        00:00:00 kworker/u2:2-events_unbound
>   103 ?        00:00:01 systemd-udevd
>   114 ?        00:00:00 SMIO
>   121 ?        00:00:00 mmal-vchiq
>   122 ?        00:00:00 mmal-vchiq
>   124 ?        00:00:00 mmal-vchiq
>   131 ?        00:00:00 mmal-vchiq
>   132 ?        00:00:00 mmal-vchiq
>   133 ?        00:00:00 mmal-vchiq
>   157 ?        00:00:00 cfg80211
>   238 ?        00:00:00 cron
>   242 ?        00:00:00 rsyslogd
>   246 ?        00:00:00 avahi-daemon
>   259 ?        00:00:00 dbus-daemon
>   261 ?        00:00:00 avahi-daemon
>   268 ?        00:00:00 wpa_supplicant
>   275 ?        00:00:00 systemd-logind
>   280 ?        00:00:00 alsactl
>   291 ?        00:00:00 thd
>   311 ?        00:00:00 rngd
[... 61 lines cut ...]
```

So the sync of time between RTC (the PiJuice) and the system failed.
Then the script attempts to set the alarm on the PiJuice, which fails 4 times, and then finally works.... so one minute the PiJuice is broken and the next minute it's working again.

> pi@raspberrypi:~/timelapse $ sudo hwclock -r
> hwclock: Cannot access the Hardware Clock via any known method.
> hwclock: Use the --verbose option to see the details of our search for an access method.

I run `i2cdetect -y 1` again and 68 has now been replaced with UU... I didn't touch anything that would alter that.

> pi@raspberrypi:~/timelapse $ i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --

I run pijuice_cli and it still recognises the pijuice is connected as I can see the charge level.

I'm beginning to think I have a flaky/faulty PiJuice...

tvoverbeek, 2020-09-13:
Something is fishy with the RTC part of the PiJuice firmware.
Suggest to reflash the firmware
The firmware is located in `/usr/share/pijuice/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary`.
You can reflash with the command `pijuiceboot 14 usr/share/pijuice/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary`
It will produce a lot of output about erasing and flashing all pages.
After flashing the firmware should restart. Check all settings because only the non-volatile ones will be preserved.
See also https://github.com/PiSupply/PiJuice/tree/master/Firmware where you will find all previous firmwares and flashing instructions like above.

See if this helps or if the errors still occur.

Note: If you start a script from cron @reboot it might execute too early before the i2c bus is initialized and then the hwclock --hctosys does not work. (Have run into this myself).

f2calv, 2020-09-14:
So I flashed the firmware then I also went into pijuice_cli and manually reset to 'default settings' - I set the battery profile and existed the tool. Note: This time I left id-eeprom at 0x50.

> pi@raspberrypi:~/timelapse/logs $ pijuiceboot 14 /usr/share/pijuice/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary
> using address 14
> Input file /usr/share/pijuice/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary
> Starting bootloader
> Starting bootloader old firmware
> Sending get command
> bytesRead 13  dataLen 11
> ack 79
> data: b, 10, 92
> data received 11
> bootloader version: b
> commands: 10, 0, 1, 2, 11, 21, 31, 44, 63, 73
> erase page count 37
> first page erase succcess 0
> erase success 0
> page count 294
> mem data read 256, 256
> Page 293 programmed successfully
> mem data read 256, 256
> 
> ...snip...
> 
> Page 2 programmed successfully
> mem data read 256, 256
> Page 1 programmed successfully
> mem data read 256, 256
> Page 0 programmed successfully
> EEPROM programming finished successfully
> Code executed successfully

I set the system clock sync and sync'd that time with the RTC and ran my app which did reboots normally until sunset and then the pi shutdown overnight as expected. It woke-up at sunrise, ran for up 1 reboot then failed when accessing `pijuice.status.GetChargeLevel()`

f2calv, 2020-09-14:
Starting a new debug process from absolute beginning on a Pi 2B this time without installing any other python packages, running any apt-get updates or other customisations;

- built fresh Raspbian install with Etcher using 2020-08-20-raspios-buster-armhf-lite.zip
- copied wpa_supplicant.conf and SSH to the boot partiton for headless setup
- logged in as pi, changed pi password with `passwd`
- ran `sudo apt-get install pijuice-base -y`
- ran `pijuice_cli` (to check status it correctly says 'unable to connect to device')
- ran `sudo poweroff`
- physically installed the pijuice, and then added the battery
- power on pi & logon
- ran `lsmod|grep i2c`
> i2c_bcm2835            16384  0
> i2c_dev                16384  0
- ran `pijuice_cli`, reset settings to defaults
- set battery profile to BP7X_1820 and applied that setting
- re-check status to see if it sees the pi battery (it did), apply settings and then exited
- ran `sudo shutdown -r now` (a reboot just to be sure)
- ran `pijuice_cli` again, tried 'status' again screen said -1% battery for a few seconds, then refreshed to show 30%
- ran date commands, and got errors from hwclock;
> pi@raspberrypi:~ $ sudo date
> Mon 14 Sep 12:25:10 BST 2020
> pi@raspberrypi:~ $ # check RTC datetime
> pi@raspberrypi:~ $ sudo hwclock -r
> hwclock: Cannot access the Hardware Clock via any known method.
> hwclock: Use the --verbose option to see the details of our search for an access method.

-ran `i2cdetect -y 1`

> 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- [...]

- assume RTC module not loaded, so load it via running `sudo nano /boot/config.txt`, adding `dtoverlay=i2c-rtc,ds1339` at the end of the file then `sudo shutdown -r now` to reboot.
- ran `i2cdetect -y 1` still came back with 68
- rebooted again, ran `i2cdetect -y 1` came back with UU this time
- ran date functions - hwclock threw a weird I/O error first time, then completed;

> pi@raspberrypi:~ $ sudo date
> Mon 14 Sep 12:31:09 BST 2020
> pi@raspberrypi:~ $ # check RTC datetime
> pi@raspberrypi:~ $ sudo hwclock -r
> 2020-09-14 12:31:11.158930+01:00
> pi@raspberrypi:~ $ # sync system datetime to RTC
> pi@raspberrypi:~ $ sudo hwclock --systohc
> hwclock: ioctl(RTC_SET_TIME) to /dev/rtc0 to set the time failed: Remote I/O error
> pi@raspberrypi:~ $ sudo hwclock --systohc
> pi@raspberrypi:~ $ sudo hwclock --systohc

- setup github.py script, `mkdir timelapse`
  - `pscp ********\timelapsepi\*.py pi@pi2-wifi:/home/pi/timelapse/` (copy files from development PC with Putty)
  - `chmod +x github.py`
- ran the script first time - **success**;
pi@raspberrypi:~/timelapse $ python3 github.py
> Checking for /dev/i2c-1... found /dev/i2c-1 in 1 attempt(s)
> Connecting to PiJuice... connection established :)
> Synchronize Pi hardware clock... CompletedProcess(args=['sudo', 'hwclock', '--hctosys'], returncode=0, stdout=b'', stderr=b'')
> GetChargeLevel();
> {'data': 26, 'error': 'NO_ERROR'}
> 2020-09-14 11:36:12 taking photo and doing stuff here (sleeping for 20 seconds)
> rebootLoopEnabled=False
> ClearAlarmFlag()
> {'error': 'NO_ERROR'}
> SetWakeupEnabled(False)
> {'error': 'NO_ERROR'}
> 2020-09-14 12:36:32 script completed in 20.43 second(s)
> sudo shutdown -h now <--DISABLED AS TESTING

- ran the script second time - **failure**;
> pi@raspberrypi:~/timelapse $ python3 github.py
[... 121 lines cut ...]
                attempt=attempt, alarm=str(alarm)))
            return True
    return False

def getAlarm():
    # set alarm for 2 minutes in the future
    DELTA_MIN = 2

    utcNow = datetime.utcnow()
    # Set RTC alarm DELTA_MIN minutes from now
    # RTC is kept in UTC
    alarm = {}
    alarm['year'] = 'EVERY_YEAR'
    alarm['month'] = 'EVERY_MONTH'
    alarm['day'] = 'EVERY_DAY'
    alarm['hour'] = 'EVERY_HOUR'
    alarm['minute'] = (utcNow.minute + DELTA_MIN) % 60
    alarm['second'] = 0

    return alarm

if __name__ == "__main__":
    init()
    main()
```

tvoverbeek, 2020-09-14:
I am running out of suggestions.
How do you power the pi/PiJuice combination now? Solar Array?
Does it do a planned shutdown at sunset? How does it start in the morning?

f2calv, 2020-09-14:
Majority of my testing it's been running off battery, now I'm testing running off a PSU plugged into the PiJuice microUSB.
I have also now uninstalled the fake hwclock and now it seems stable, i.e. it only failed once.

It does a shutdown at sunset and the wake-up alarm is then set for sunrise, in-between sunrise/sunset it reboots every 2 minutes. It usually 'wakes up' just fine at the sunrise alarm, but then the script fails because of the COMMUNICATION_ERROR - lets see if it fails again tomorrow AM.

And thanks for your help so far.

f2calv, 2020-09-16:
Ok, so the good news @tvoverbeek is that all is now good - no more COMMUNICATION_ERROR thanks for your patience :)

The solution, reset everything to defaults, and don't touch any configuration options unless it's absolutely needed.

My primary mistake was reading lots of previously closed forum issues and assuming that the changes/tweaks mentioned in them would be needed for my (simple) requirements.

closed 2020-09-16 by f2calv

tvoverbeek, 2020-09-16:
@f2calv Glad you got it working. All the best with your timelapse project.