PiSupply/PiJuice #544: Using a LiFePo4 Battery

jrhelbert, 2020-09-13:
Based upon some loose information I found on here as well as options I've found in the software, it seems like I should be able to use a 1s LiFePo4 battery with the PiJuice? I am attempting to integrate this with a 6000mAh 32700 cell I have on hand, and planning to drop in a large 90Ah cell down the road for super extended runtimes.

I've attempted to set up a custom battery profile for the 6000mAh cell I have and I've run into a few issues:

1. Despite selecting LIFEPO4 as the battery chemistry, the status page reports < 10% SOC on a fully charged LiFePo4 cell. Looking at the data sheet for the fuel gauge it looks like it only supports LiIon/LiPo batteries, so I'm assuming I need to use the MCU for Rsoc?
1. Trying to use the MCU for SOC I am attempting to change the OCV10/50/90 and R10/50/90 values to match my current cell. I am able to edit the R10/50/90 values, but as soon as I try to type anything into any of the OCV fields the whole cli program crashes with the following error:
`Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2013, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 825, in _loop
    self._watch_files[fd]()
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 404, in <lambda>
    event_loop, callback, self.get_available_raw_input())
  File "/usr/lib/python3/dist-packages/urwid/raw_display.py", line 502, in parse_input
    callback(processed, processed_codes)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 411, in _update
    self.process_input(keys)
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 511, in process_input
    k = self._topmost_widget.keypress(self.screen_size, k)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 595, in keypress
    *self.calculate_padding_filler(size, True)), key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1590, in keypress
    key = self.focus.keypress(tsize, key)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2271, in keypress
    key = w.keypress((mc,) + size[1:], key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 999, in keypress
    key = focus_widget.keypress((maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 622, in keypress
    return self._original_widget.keypress(maxvals, key)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 1724, in keypress
    unhandled = Edit.keypress(self,(maxcol,),key)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 1488, in keypress
    self.insert_text(key)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 1414, in insert_text
    self.set_edit_text(result_text)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 1369, in set_edit_text
    self._emit("change", text)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 460, in _emit
    signals.emit_signal(self, name, self, *args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 265, in emit
    result |= self._call_callback(callback, user_arg, user_args, args)
  File "/usr/lib/python3/dist-packages/urwid/signals.py", line 295, in _call_callback
    return bool(callback(*args_to_pass))
  File "/usr/bin/pijuice_cli.py", line 1061, in <lambda>
    urwid.connect_signal(edit, 'change', lambda x, text, idx: self.profile_data.update({self.EDIT_KEYS[idx]: text}), i)
IndexError: list index out of range`

I'd just like to confirm I should be able to use a LiFePo4 cell with the PiJuice. It seems to be working normally and will run off the batter, and is also charging normally (regulation voltage set to 3640, and the pijuice stops charging when it should. The only issue is my SoC is completely messed up and I would like to get that working correctly.

tvoverbeek, 2020-09-13:
@jrhelbert Yes you can use single cell LiFePo4 batteries.
The MCU SOC estimation uses the OCV (Open Circuit Voltage) and battery resistance values.
The pijuice_cli crash you see is a bug. Working on a solution.
The gui version does work (if you can use the desktop version).

jrhelbert, 2020-09-13:
And I do need to use the MCU for SOC if I am using LiFePo4?

jrhelbert, 2020-09-14:
@tvoverbeek I am assuming I can set the values via one pi running a GUI and then move the pijuice to another point that is headless and the values will stay?

tvoverbeek, 2020-09-14:
Yes, the chosen battery profile is stored in non-volatile memory on the PiJuice.

jrhelbert, 2020-09-14:
Thanks, I actually managed to get a Pi running desktop just before you posted and was able to confirm that. 

I've got some rough estimates in for my OCVxx and Rxx values, and happened across the pijuice-soc-test.py script in the firmware folder. Is this still a valid script for measuring these values? 

It says to place a load resistor across the Vsys lines and uses 10ohm as an example. What's the limit for this resistor range? I happen to have a 2ohm/400W dummy load I use for capacity measurements elsewhere, but I'm not sure if that would just saturate the regulator for Vsys and invalidate the test.

jrhelbert, 2020-09-14:
I keep answering my own questions as I re-read the documentation... so I think it should work?

`The voltage for VSYS is in principle the battery voltage (nominal 3.7V, varies between 3V and 4.2V depending on charge level)`

`Max. current that the PiJuice will be able to supply
The maximum current at 5V via the GPIO is 2.5A and via VSYS is 2.1A. This is also dependent on battery capacity. For BP7X battery for example we have measured around 1.1A at 5V GPIO and around 1.6A at VSYS output. To achieve the maximum of 2.5A you will need to use a battery of at least 3500mAh.`

Looks like even at peak voltage the max current with a 2ohm resistor will be 1.8-1.9A, and shouuldn't trigger the current limiter for the Vsys connection.

tvoverbeek, 2020-09-14:
Yes, I believe it should work.
You will have to adjust the parameters for LtFePo4 (endvolt will be lower).

tvoverbeek, 2020-09-14:
@jrhelbert I just pushed a fix for the pijuice_cli crash when editing the OCV parameters.
Could you try to replace /usr/bin/pijuice_cli.py with the github version (https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_cli.py)?
Note that /usr/bin/pijuice_cli.py  has owner pijuice and group pijuice. Use chown if needed.

jrhelbert, 2020-09-14:
Well no dice of the soc_test. Prior to attaching the load I decided to double check the output behavior and I can’t get my Vsys output to toggle either manually or via the script. 

I bugged out the circuit based on the schematic and my U9 doesn’t appear to be functioning correctly. IN measures the same voltage as the battery, and i see the enable line toggling from the uProc (3.3V when pijuice_cli sets the system switch to off, 0V when set to either 500mA or 2100mA) but I’ve got nothing on the output line. Ironically the output line floats to ~20mV when system switch is set to off, but actively clamps to 0V when enabled in any way. This is true regardless of if I have my old attached or not.

Thinking it was maybe an errant current limit glitch I shorted ILIM to ground, but no change in behavior. FLAG is sitting just above 0V, which looks like the part thinks something bad is going on :/

I’ve never used Vsys before, some I’m not sure what happened to the part.

jrhelbert, 2020-09-18:
I've been looking at your notes for setting the default battery profile via R20 so that the pijuice doesn't start trying to overcharge the LiFePo4 battery in the event that nvram gets reset and it looks like Setting code 7 is probably a good safe default choice for this. Here the spreadsheet gives a value of 4600 and the column is listed as being in kOhm. So I need a 4.6 MOhm resistor for this or is that a typo and it should be 4.6 kOhm?

Also, the hardware notes state that the resistor should be 0.1% tolerance... I am having problems finding a 4.6x10^x resistors let alone anything of that tight tolerance, do you have a recommendation for what to use here?

tvoverbeek, 2020-09-19:
@jrhelbert It is a 4.6 MOhm resistor.
Note that entries 4,5, and 6 have the same termination current and regulation voltage although a lower charge current
So any resistor between e.g. 5.9 MOhm and 4.6 MOhm would be OK
You could take a lower precision resistor e.g. a 5.1 MOhm 5% resistor

jrhelbert, 2020-09-21:
Thanks,
I ordered a new PiJuice that should be here today. I'll let you know how the battery characterization goes.

jrhelbert, 2020-09-24:
Got the pijuice-soc-test script to run great with the new pijuice (43 hour test with a 2ohm load!), although I'm not sure how effective the results will be. At that high of a capacity the battery resistance is so low that it is being dominated by all my my cabling and connectors. Looks like R10 to R90 only vary about less than half a mOhm, while the rest of the setup come in at 19 mOhm, oh well. :)

I kicked off another test with the battery I intend to use in the final system (dirt cheap 90Ah LiFePo4 off Aliexpress) and I've found a limitation in the firmware - I can't enter a capacity above 65,535 mAh.

I'm guessing it would be more work than it's worth to increase that field size?

jrhelbert, 2020-09-24:
> @jrhelbert I just pushed a fix for the pijuice_cli crash when editing the OCV parameters.
> Could you try to replace /usr/bin/pijuice_cli.py with the github version [...]

I also finally tested out the new pijuice_cli you pushed, I can confirm that I can edit the OCV fields now from the CLI

tvoverbeek, 2020-09-24:
@jrhelbert In the firmware the battery capacity is stored in a uint16 (unsigned 16-bit integer) hence the max value of 65535.
Larger capacities would require a rebuild of the firmware with capacity as an uint32 + associated changes everywhere where the capacity is used.
You would have to address this directly to Pi-Supply (@shawaj @ChristopherRush ) to let the firmware developer assess if this can be implemented.

closed 2021-06-01 by jrhelbert