PiSupply/PiJuice #545: Command line query

tlewis99, 2020-09-15:
I have over 70 units in production and want to know is there an option to query from the command line to find the pijuice hat's current firmware and current battery profile?

Thanks,

Tim

tvoverbeek, 2020-09-16:
@tlewis99 Here a python script which provides the requested information:
```
#!/usr/bin/python3

from pijuice import PiJuice
pj = PiJuice(1,0x14)

ret = pj.config.GetFirmwareVersion()
if ret['error'] == 'NO_ERROR':
    print('Firmware version = ', ret['data']['version']
else:
    print(ret['error'])

ret = pj.config.GetBatteryProfileStatus()
if ret['error'] == 'NO_ERROR':
    print('Battery profile = ', ret['data']['profile'])
else:
    print([ret['error'])
```

tlewis99, 2020-09-16:
I'm getting a syntax error and my python developer is on extended vacation so I'm lost

pi@TS999:/home/tech/scripts $ /usr/bin/python3 check_pi_setup.py
  File "check_pi_setup.py", line 9
    else:
       ^
SyntaxError: invalid syntax

tvoverbeek, 2020-09-16:
My bad.
On line 8 (`    print('Firmware version = ', ret['data']['version']`) the closing parentesid of the print statement is missing.
It should be:
`    print('Firmware version = ', ret['data']['version'])` <===

tlewis99, 2020-09-16:
Thanks, but not quite there yet:
Here is the script:
#!/usr/bin/python3

from pijuice import PiJuice
pj = PiJuice(1,0x14)

ret = pj.config.GetFirmwareVersion()
if ret['error'] == 'NO_ERROR':
    print('Firmware version = ', ret['data']['version'])
else:
    print(ret['error'])

ret = pj.config.GetBatteryProfileStatus()
if ret['error'] == 'NO_ERROR':
    print('Battery profile = ', ret['data']['profile'])
else:
    print([ret['error'])

And here is the current error:
tech@TS999:~/scripts $ /usr/bin/python3 check_pi_setup.py
  File "check_pi_setup.py", line 16
    print([ret['error'])
                       ^
SyntaxError: invalid syntax

tvoverbeek, 2020-09-16:
OK. Too many typos.
Line 16 `print([ret['error'])` should be `print(ret['error'])`
Delete `[` after `print(`
Hope this was the last one ;-).

tlewis99, 2020-09-16:
That works thanks!

I assume it can only be run at the user pi

tlewis99, 2020-09-16:
Below is the final code if someone needs it:
`#!/usr/bin/python3

from pijuice import PiJuice
pj = PiJuice(1,0x14)

ret = pj.config.GetFirmwareVersion()
if ret['error'] == 'NO_ERROR':
    print('Firmware version = ', ret['data']['version'])
else:
    print(ret['error'])

ret = pj.config.GetBatteryProfileStatus()
if ret['error'] == 'NO_ERROR':
    print('Battery profile = ', ret['data']['profile'])
else:
    print(ret['error'])
`
I called the python script:
check_pj_setup.py
and I run it like this from the same directory as I have created the script:
sudo -u pi ./check_pj_setup.py

tvoverbeek thanks for this and I am closing this issue.

closed 2020-09-16 by tlewis99

tvoverbeek, 2020-09-17:
@tlewis99 It can be run by any user who belongs to the i2c group (i.e. has access to the I2C bus)