PiSupply/PiJuice #546: temperature being reported that of board, not battery

KLForsythe, 2020-09-17:
Battery: PiJuice 2500mAh
Battery Profile selected: SNN5843_2300 
Temperature sensor selected: NTC  (`pijuice.config.SetBatteryTempSenseConfig('NTC')`)

GetBatteryTempSenseConfig() reports: {'error': 'NO_ERROR', 'data': 'NTC'}

However, the temperature being reported is not the NTC value, but the temperature of the board.  The battery is cool to the touch, however, warm/hot temperatures are being reported.  

The issue is not the battery itself; a battery replacement did not fix the problem.  

The specific temperatures over the first 5 minutes after boot with the replacement battery were as follows:

At 12:12pm, the battery reported 26C temp and 16% level
At 12:14pm, 46C, 19%
**At 12:16pm, I used the cli to switch from NTC to on board temp readings, and saw 47C/21%**
At 12:17, having re-selected NTC temp, 48C/22%

Throughout, the battery was cool to the touch.  As noted, when switching sensor from NTC to on-board, the temperatures reported were basically the same.  

After significant investigation via a help ticket, I was advised to open a github issue.  

If additional temperature readings would be helpful, I can provide either further temp readings from discussion in the help ticket mentioned, or any new readings that could help. 

I'd greatly appreciate any assistance in resolving this problem.  Thank you so much!

tvoverbeek, 2020-09-17:
@KLForsythe Is this with the PiJuice Zero or the PiJuice HAT.
It smells like the NTC lead is not connected. (Firmware falls back to board temperature if it cannot read the NTC)

KLForsythe, 2020-09-17:
@tvoverbeek This is the PiJuice HAT.  

The NTC lead is connected.  @ChristopherRush originally thought the connection made by twisting the wires might be insufficient, and recommended soldering the wires, which we did. Unfortunately, that did not fix the problem. 

So the NTC wire should have a good connection. 

[image]

tvoverbeek, 2020-09-17:
Since there is no battery profile in the firmware for the 2500 mAh battery there is the possibility the NTC parameters
(Resistance and Beta factor) used in the 2300 mAh profile do not match what is needed for the 2500 mAh one.
Just for a test you could make a Custom profile starting from the 2300 mAh profile and only change the Beta (NTC B constant). It is 3380 for the 2300 mAh, Change it to e.g. 3450 and see if it changes the reported battery temperature.
If it does deviate from the board temperature with a changed NTC B constant then we know the parameters do not match the 2500 mAh.
You will probably need to contact Pi-Supply again to get the real values for the 2500mAh one.

@ChristopherRush @shawaj Do you have the profile parameters for the 2500mAh battery? (PS the picture on the shop website for the 2500mAh is in fact the 5000 mAh one)

KLForsythe, 2020-09-17:
@tvoverbeek I changed the NTC B constant to 3450 as recommended, but the reported battery temperature did not change:

```
>>> from pijuice import PiJuice
>>> pijuice = PiJuice(1, 0x14)
>>> battery_profile_args = {'capacity': 2500, 'chargeCurrent': 1125, 'terminationCurrent': 100, 'regulationVoltage': 4180, 'cutoffVoltage': 3000, 'tempCold': 1, 'tempCool': 10, 'tempWarm': 45, 'tempHot': 59, 'ntcB': 3450, 'ntcResistance': 10000}
>>> pijuice.config.SetCustomBatteryProfile(battery_profile_args)
{'error': 'NO_ERROR'}
>>> pijuice.config.SetBatteryProfile('CUSTOM')
{'error': 'NO_ERROR'}
>>> pijuice.config.GetBatteryProfile()
{'data': {'tempCold': 1, 'terminationCurrent': 100, 'chargeCurrent': 1075, 'capacity': 2500, 'ntcResistance': 10000, 'tempHot': 59, 'regulationVoltage': 4180, 'tempWarm': 45, 'ntcB': 3450, 'cutoffVoltage': 3000, 'tempCool': 10}, 'error': 'NO_ERROR'}
>>> pijuice.config.GetBatteryTempSenseConfig()
{'data': 'NTC', 'error': 'NO_ERROR'}
>>> pijuice.status.GetBatteryTemperature()
{'data': 52, 'error': 'NO_ERROR'}
>>> pijuice.config.SetBatteryTempSenseConfig('ON_BOARD')
{'error': 'NO_ERROR'}
>>> pijuice.status.GetBatteryTemperature()
{'data': 51, 'error': 'NO_ERROR'}
>>> pijuice.config.SetBatteryTempSenseConfig('NTC')
{'error': 'NO_ERROR'}
>>> pijuice.status.GetBatteryTemperature()
{'data': 51, 'error': 'NO_ERROR'}
```

tvoverbeek, 2020-09-17:
I do not have a 2500 mAh battery myself. Maybe it does not have a NTC internally. That is what it smells like.
Suggest to contact Pi-Supply again with reference to this issue.
Hope you get it solved.

KLForsythe, 2020-09-17:
@tvoverbeek 

@ChristopherRush does have a 2500 mAh battery, and he said he tested his and was getting accurate NTC readings when he tested his.  Based on that, I think the 2400 mAh battery must have a NTC.  

Additionally, I have a different 2500 mAh battery running on the same OS and the same software, and it _usually_ reports NTC values, although sometimes it seems to pick up on-board temp instead for a brief time.  

However, the one I reported on has consistently just reported on-board temps.

tvoverbeek, 2020-09-17:
Something else. Which version of the firmware are you using?
The previous version (1.3) was reporting fluctuating temperatures.
Anyway, then it seems your problematic may have a faulty NTC.

KLForsythe, 2020-09-17:
I am using firmware version 1.4 (software version 1.6) 

@ChristopherRush had the same idea about a faulty NTC in the battery, and got me a replacement battery.  The replacement had the same problem as the original.

tvoverbeek, 2020-09-17:
I sm out of ideas. 1.4 is the current version.
@ChristopherRush @shawaj Could it be that the whole 2500 mAh lot at Pi-Supply has no NTC?

ChristopherRush, 2020-09-21:
@tvoverbeek @KLForsythe I have just come across this issue whilst doing some testing. When I switch from NTC or Auto to on-board temperature sense, temperature results are as normal 23c NTC and 40C on-board. When I switch back to NTC temperature reported temperature is 23c but then after a few minutes switches to 40c and I'm no longer able to get the NTC temperature value back. The only way to restore the settings is to re-connect the battery.

KLForsythe, 2020-09-24:
@tvoverbeek Any thoughts on how to resolve this?  Now that @ChristopherRush has been able to replicate the problem, I'm hoping a solution will be possible.

tvoverbeek, 2020-09-24:
@KLForsythe This is most likely a firmware issue. The NTC temperature is read via the fuel-gauge IC.
@ChristopherRush Can you ask @mmilann (the firmware author) to look into this?

KLForsythe, 2020-09-28:
@mmilann, if you are able to look into this, I'd greatly appreciate it.  Any updates or information you can provide would also be appreciated.  

We are hoping to release a product which uses the PiJuice very shortly, but need to be able to get correct NTC temperature readings before we can release.  

I'd be happy to provide any information or assistance I can.  

Thanks again.

mmilann, 2020-09-30:
@KLForsythe Can you check readings when battery is heated over 25C? 

There is safety threshold temperature of 45C, so when board heats above 45C and battery temp is between 23-27C (which may mean that battery have incorrect of fake NTC sensor), Pijuice uses on-board temperature. Batteries with fake NTC or just 10Kohm resistor instead of NTC always reports around 25C.

Purpose of battery temperature measurement is for safety like to disable charging when it is too hot (configurable via battery profile). If there is no NTC detected it uses on-board readings. If battery is cable connected difference between board temperature and battery may be greater, especially in Rpi 4 case and when charging board becomes hot.

KLForsythe, 2020-09-30:
@mmilann You are correct.  When the battery is warmer, the NTC values are indeed reported correctly.

So we were receiving the on-board temperatures because the board was over 45C, and the battery was between 23 and 27C.  From your explanation, I understand that even though we had a valid NTC, it was assumed that it was inaccurate because our board was warm and the battery was cool.  

Thanks for the explanation.  I'll do some additional testing, but I don't believe we will encounter this situation outside of lab testing, since with our battery in the same case as the board (connected by cable, and positioned a couple inches away from the board), it is consistently warmer than 27C.

tvoverbeek, 2020-09-30:
@KLForsythe 
Just confirming @mmilann's explanation.
In battery.c in the firmware source code:
If (23 <= ntc temperature <= 27) and (board temperature < 15 or board temperature > 45) then switch to board temperature measurement.
Note the 45 board temperature limit is independent from the warm temperature limit in the battery profile.

KLForsythe, 2020-09-30:
@mmilann @tvoverbeek Is there a way to override this?  

On additional testing, there are situations in which our board is >45C, and our battery is between 23 and 27 NTC temp.  We are using rPi 3B+ boards, but we do have a lot running on them.  But these are well-ventilated cases, with the battery connected by cable, and several inches away from the board.  We set them up this way to keep the battery as cool as possible, as when we were using the BP7X_1820 batteries, we had considerable problems with the battery overheating.  So we switched to these batteries https://uk.pi-supply.com/products/pijuice-2500mah-battery and designed a case that would provide additional ventilation and hold the battery well off the board.  

Now, it seems our batteries are running too cool compared to the board temp, so the firmware is defaulting us to the onboard temp, which is considerably different from the NTC battery temp.   This is a problem, because in addition to reporting a temperature that isn't reflective of the actual battery temperature (NTC), it means that when the board is warm, charging will be slowed or disabled, even if the battery itself is at a good temperature for charging.

tvoverbeek, 2020-09-30:
@KLForsythe Override directly, No. This would need a new firmware.
But you can mitigate a bit.

The board temperature is the PiJuice board temperature, not the RPi board temperature.
Getting the PiJuice further away from the 3B+ should lower the PiJuice board temperature.

There are only a few connections needed between the RPi and the PiJuice:
- 5V pins 1 or 3
- I2C pins 4(SDA) and 5(SCL)
- GND any of pins 5,13,19,29,33,10,26,40
If you do rely on the ID-EEPROM you also need:
- pin 28 (ID_SDA) and pin 27 (ID_SCL)
- 3V3 pin 18

Connect the PiJuice board away from the Pi3B+ with only the wires you need.
Or you could use a stacking header to get extra distance between the Pi3B and the PiJuice.
Do not know if your case design is final already.

Also (but this is a bit cheating) change the battery profile warm temperature a bit above 45 (using the custom profile starting
from your current profile), so the charging reduction will start at a higher temperature.

KLForsythe, 2020-09-30:
@tvoverbeek @mmilann 

In observing the temperature readings once the firmware starts overriding the NTC temp with the on-board temperature, once the firmware starts overriding the temperature, it does not seem to ever re-check the NTC temp.   (which would be consistent with what @ChristopherRush observed here: https://github.com/PiSupply/PiJuice/issues/546#issuecomment-696149559).  Within 10-30 minutes of turning the system on, the battery itself (and therefore NTC temp) should be above 27C.  But the reported temperature never switches back to NTC once the NTC temp is what the firmware considers an acceptable value.   

Is there any way to prompt a re-check, so the temperature reading can switch back to NTC once it is reporting within the acceptable range (> 27C)? 

Once our system has been on for 10-30 minutes, our NTC readings usually are in the 31-39 range, and stay that in that range whether charging, not charging, or discharging. So, if we could prompt a recheck of the NTC value, it would be within range at that point. 

@tvoverbeek  regarding your suggestions:

>Connect the PiJuice board away from the Pi3B+ with only the wires you need.
Or you could use a stacking header to get extra distance between the Pi3B and the PiJuice.

This would require a redesign of our case.  Which, while not impossible, would be difficult at this stage. 

>Also (but this is a bit cheating) change the battery profile warm temperature a bit above 45 (using the custom profile starting from your current profile), so the charging reduction will start at a higher temperature.

This might mitigate the slow charging, but we'd still get the wrong temperature readings.  

I'm currently testing some heat sinks, to see if they keep the PiJuice board temperature low enough for long enough to avoid the battery still being too cool when the PiJuice board reaches 45C.

tvoverbeek, 2020-09-30:
I do not know if this helps, but after 10-30 minutes you could force a reset of the PiJuice: `pijuice.config.SetDefaultConfiguration()`.
Disadvantage: this resets the battery profile and all other volatile settings to their default values,
so you would have to reprogram the battery profile and any other settings you need.

KLForsythe, 2020-09-30:
@tvoverbeek Thanks!  I'll try that out.

zt4, 2021-12-27:
Are there any plans with a new firmware to have the ability to disable this safety threshold board temperature of 45C?

I have an external battery and reporting the board temperature is useless and is the opposite of safe, as I have no way of telling the actual battery temperature. And as mentioned above, it appears to get stuck on the board temperature, so even when I put a fan on the board to cool it down, I don't get the battery temperature back again.

This could just be another option under the Temperature Sense Menu: NOT_USED, NTC, ON_BOARD, AUTO_DETECT, **FORCE_NTC**

Thanks!

tvoverbeek, 2021-12-27:
The 45C limit is in the battery profile.
If your battery does not have a NTC then use NOT_USED. This forces  a default temperature of 25C.
If your battery does have a NTC you need the NTC characteristics to be put in the battery profile (you will need a custom profile).

zt4, 2021-12-27:
> The 45C limit is in the battery profile. If your battery does not have a NTC then use NOT_USED. This forces a default temperature of 25C. If your battery does have a NTC you need the NTC characteristics to be put in the battery profile (you will need a custom profile).

I am using the PiJuice 12000mAh battery. The issue is the comment https://github.com/PiSupply/PiJuice/issues/546#issuecomment-701286737 where the battery board is above 45C and the battery is between 23-27C.

`There is safety threshold temperature of 45C, so when board heats above 45C and battery temp is between 23-27C (which may mean that battery have incorrect of fake NTC sensor), Pijuice uses on-board temperature. `

However, it also gets stuck reading the board temperature (as noted in comment https://github.com/PiSupply/PiJuice/issues/546#issuecomment-701535482 it may be after being above 45C for 10+ minutes). So if I cool the board down below 45 (with a fan), I still cannot read the battery temperature, just the board temperature.

As this is an external battery, I always want to use the NTC, never the board temperature. I understand the reason for including the safety 45C board temperature, but with an external battery, that just causes issues with the ability to read the actual battery temperature. So I was wondering if a new FORCE_NTC option could be added to the Temperature Sense menu for cases when you know you are bypassing the board temperature safety protection.