PiSupply/PiJuice #547: PiJuice Zero buttons not functioning

peterallennn, 2020-09-18:
I have the SW1 button on the pijuice zero board configured like so:

[image]

But no matter what I do, I cannot get the thing to power off.

The pijuice service is active:

[image]

I've tried uninstalling and reinstalling the pijuice-base package.

Can anyone assist with this please?
Is there a log file available that records when the button is pressed and whether there were any errors?

Thanks in advance.

tvoverbeek, 2020-09-18:
Note you also have "Single Press" defined.
Single press is registered if you release the button less than 0.8 seconds after you press it.
Single Press will definitely override your Press event. 
Try assigning Single Press to No Function and see if you get your hard power off (Not a good thing to do if your Pi is on battery power) .

peterallennn, 2020-09-18:
Aha, the functions are now working.

Okay now the button seems to be doing something, on the double press action, I am trying to run a user script to simply blink the D2 LED and write a test.txt file in the /home/pi directory and then connect/disconnect to wifi.

I run the below code with Python3 and it functions as expected. However, with the double press user script, the LED blinks but the test.txt file isn't created nor the wifi connected/disconnected. Is this because of permission issues? 

[image]

tvoverbeek, 2020-09-18:
User scripts have to be executable and be owned by a non-privileged user (e.g. pi) which belongs to the pijuice group.
Since /etc/wpa_supplicant/wpa_supplicant.conf is owned by root and can only be read/written by root your wpa_supplicant.conf changes will not work.
 'iwgetid -r' does not need to run as privileged user and I do not believe you need the systemctl daemon-reload (only when service definitions are changed/deleted/added)

shawaj, 2020-10-24:
@peterallennn is this all working for you now?

peterallennn, 2020-10-24:
@shawaj Yep thanks Aaron and thanks to @tvoverbeek for the help.

shawaj, 2020-10-24:
@peterallennn perfect, will close issue now then

closed 2020-10-24 by shawaj