PiSupply/PiJuice #548: reboot and I2c is not on

JoshYang1977, 2020-09-21:
Hello,

I am working on a project including a pi-zero, PiJuice Zero, and a KS-103 ultrasonic sensor to measure distance.
Thanks to all wonderful feeds, I was able to turn off/on the system on schedule. However, I ran into a problem.
The I2C KS-103 device seems not working (the led on KS-103 is not blinking) every time when the system restarts.

The whole thing works at first time but does not work after the system restarts on schedule.
Thank you very much for your response.

Josh

tvoverbeek, 2020-09-21:
As far as I know the KS-103 only blinks its I2C address when it is powered up.
Are you sure the power to the KS-103 is removed when you shut down the system?
A normal shutdown on the Pi (e.g. by `sudo poweroff`) does not remove the 5V and 3.3V power on the GPIO pins.
How do you power the pi-zero,pijuice-zero combination? Via the pi-zero micro-usb or via the pijucie-zero micro-usb?

JoshYang1977, 2020-09-22:
Thank you very much for your quick response.
Yes, I noticed the power is not to the KS-103. I tried both the power supply locations:  pi-zero micro-usb and  the pijucie-zero micro-usb. It returned the same result to me. Working for the first time but not working after restart the system.

Finally, I found out the way I assigned the I2C address is outside the subroutine. In consequence, it would not wake up the KS-103 while restarting the system. It works when I change the way I assign the address of I2C for the KS-103.

Thank you very much and I shall close the issue. Very much appreciate your help.

JoshYang1977, 2020-09-22:
> As far as I know the KS-103 only blinks its I2C address when it is powered up.
> Are you sure the power to the KS-103 is removed when you shut down the system?
> A normal shutdown on the Pi (e.g. by [...]

A quick question: if I power the KS-103 from different sources, does it have an impact on the power supply when the system is restarting?

JoshYang1977, 2020-09-22:
> As far as I know the KS-103 only blinks its I2C address when it is powered up.
> Are you sure the power to the KS-103 is removed when you shut down the system?
> A normal shutdown on the Pi (e.g. by [...]

tvoverbeek, 2020-09-22:
It depends if the KS-103 is then back-powering the Pi via the GPIO. If the supply is completely independent it could work.

JoshYang1977, 2020-09-22:
Thank you very much for your reply.

The most current update:
It seems work if I power via Pi Zero micro-usb, the whole thing will restart and the KS-103 will do the measuring task.
However, it just stopped after doing about 10 times ( the "power off" and "repeat" interval is 10 mins).

If I power via PiJuice micro-usb, it did not work at all. The system will restart on schedule but the KS-103 just does not have power. 

Once again, thank you very much for your thought and idea.

shawaj, 2020-10-24:
@JoshYang1977 closing issue now as no activity in a while. Feel free to reply if you do not agree

closed 2020-10-24 by shawaj