PiSupply/PiJuice #549: No issue, just a question

lotte67890, 2020-09-25:
I´m thinking of using a a Pi3 or 4 for car multimedia, would the SW for the PiJuice run on libreelec? All I would need is to power up on DC in and Power down when DC goes off.

Thanks

L

ChristopherRush, 2020-09-25:
I belive that LibreElec is read only file system as well as missing a lot
of dependancies that pijyuce requires.

tvoverbeek, 2020-09-25:
As @ChristopherRush wrote, probably a lot of dependencies are missing. LibreElec is by definition 'just enough os for Kodi'.
I do not even know if Python3 is included in LibreElec.
Anyway, power up ca be done without software support.
For power off you will need the pijuice service running to react on the 'no power' event.
THis requires Python and OS support.
You are probably better off running Kodi on top of Buster Lite.

lotte67890, 2020-09-26:
Thanks a lot, I will give it a try!

lotte67890, 2020-09-28:
I tried a lot on the weekend and all requirements I need (except from the PiJuice) seem to be solved :-). One thing I need to know is if I use an exterenal SSD powered through the USB Port on the PI(4B) is the Power supply from the PiJuice sufficient to satisfy the 2W (0,4A) needed from the SSD? I ask since I read that the Battery from the PiJuice runs with 3,7V and USB uses 5V. I do not need a long buffer time from the Battery, a couple of minutes should be enough.

Thank you

L

tvoverbeek, 2020-09-28:
@lotte67890 Depends on the total current needed by the Pi4 + SSD combination.
The default BP7X battery could be marginal. I suggest to use a larger capacity battery to be on the safe side.
There are 2 DC-DC converters involved: one from battery or external power to system power and one from system power to 5V.

lotte67890, 2020-09-28:
Thank you! As far as I read the Pi4B takes max 7,6W and the SSD 2W which makes a total of less than 10W. How much mAh has the BP7X? Which model would you suggest? 
And by the way, do you know a dealer in Germany or wherever were I can order one set so it may reach me before the end of the week? :-)

shawaj, 2020-09-28:
@lotte67890 Distrelec, Digi-key, Mouser and a few others should skip to Germany very quickly

lotte67890, 2020-09-28:
@shawaj Thank you, Digi-key is just around the corner... didn´t know them, always thought Conrad is the only one around and they don´t have it...

shawaj, 2020-09-28:
Digi-Key is a USA company, but I think Munich is their EU base :-)

lotte67890, 2020-10-12:
So, thanks for your help so far, I received the PiJuice in time. Anyway I´m facing some issues. I set the delay Power down time to 30 seconds but the shutdown after the power is disconnected occurs with a ~10 secs delay... Maybe I have sort of a missunderstanding of the setting.
Also the RTC can´t be accessed after a couple of days. I followed the steps as adviced in #477 and get:
```
pi@raspberrypi:~ $ grep i2c /boot/config.txt
dtparam=i2c_arm=on
pi@raspberrypi:~ $ lsmod | grep i2c
i2c_bcm2835            16384  2
i2c_dev                20480  4
pi@raspberrypi:~ $
```
but i2cdetect -y 1 says 68 instead of UU. And cause of that is that I can not use hwclock -s. I have no clue what should have  changed at the system and it worked in the beginning out of the box. I also reset the Pijuice settings back to defaults with no luck... It´s not that important for my setup, but I wonder why it´s not working....

tvoverbeek, 2020-10-12:
@lotte67890 You wrote:
```
I set the delay Power down time to 30 seconds but the shutdown after the power is disconnected
occurs with a ~10 secs delay... Maybe I have sort of a misunderstanding of the setting.
```
The 'Software Halt Power Off' delay has nothing to do with the reaction on the 'no power' event.
'Software Halt Power Off' is used when the shutdown is commanded by a user program on the RPi.
When on battery only the 5V supply for the RPi from the PiJuice is switched off after the specified
delay time (delay is started when the shutdown is commanded, so you need sufficient margin to be sure
the shutdown sequence has completed before the 5V power disappears).
The 'no power' event (when enabled) is detected with a delay after the external power (to both the RPi and the PiJuice)
disappears. This to guard against a short power interruption.
The ~10 sec delay you mention is to  when the shutdown sequence completes?

For the RTC, what steps exactly did you take from #477 ?
Anyway 68 means there is no linux driver loaded for the RTC.
Do you have any lines in /boot/config.txt for the RTC?
What is the ID-EEPROM address specified for the PiJuice (0x50 or 0x52)?

lotte67890, 2020-10-12:
Dear tvoverbeek, thanks for explaining the different "Power off" meanings.
With the ~10 sec delay I mean the time when the Pijuice initiates the shutdown sequence after loss of power input.

Regarding the RTC, the ouput I posted is exactly what I get, although I copied the lines from #477, so in my case it is:
```
pi@luigi:~ $ grep i2c /boot/config.txt
dtparam=i2c_arm=on
pi@luigi:~ $ lsmod | grep i2c
i2c_bcm2835            16384  2
i2c_dev                20480  4
pi@luigi:~ $
```
The ID-EEPROM address is 0x50
```
pi@luigi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

```

tvoverbeek, 2020-10-12:
How do you power the Pi/PiJuice combination?
If you start with applying power to the Pi, the Pi will start booting and also the microprocessor on the PiJuice will start.
However the PiJuice firmware needs time to initialize before it can answer I2C requests.
The Pi RTC init might then be too early and will fail (-> no driver loaded -> later i2cdetect will show 68 instead of UU).

lotte67890, 2020-10-12:
Ah, I think now I know what the issue might be. I switched the boot device from SD to SSD. The OS now boots in aprox 5 secs. Anyway the power is connected to pijuice.
Do you think rmmod i2c_bcm2835 & modprobe i2c_bcm2835 in a startup script could do the trick?

tvoverbeek, 2020-10-12:
The i2c_bcm2835 module is not the problem.
Somehow the rtc module is not loading.
Do you get any output from `lsmod|grep rtc`?
You could try the following:
Set the ID-EEPROM address to 0x52 (-> the dtoverlay in the eeprom will not be used).
In /boot/config.txt add `dtoverlay=i2c-rtc,ds1307`. This lets the OS use the clock function but not the alarm.
The alarm can still be programmed via the Python API.

lotte67890, 2020-10-13:
Ok, that worked. Thanks for that! It even works regardless if I switch the EEPROM to 50 or 52.
But... I now have a script running as User shutdown event, which I need to grab some data before I switch off the pi with "sudo halt" (Raspi OS Buster). This works fine except the fact that the pi does not power up when dc is back on the pijuice... Did I miss something?
Thanks
Lothar

tvoverbeek, 2020-10-13:
Yes, If you shut down per user script you also have to enable wakeup-on-charge again. The flag in the firmware gets cleared
after it wakes up the Pi. If you use the SYSTEM_XXX functions to shut down the pijuice service does this for you.
Just add `pijuice.power.setWakeUpOnCharge(0)` just before you do `sudo halt` in your script.
Then the Pi should boot when dc comes back.

lotte67890, 2020-10-13:
I see, but how exactly do I have to specify the command? I use a bash script and guess I need to run it with one of your python commands...

lotte67890, 2020-10-13:
Memo to myself: https://learn.pi-supply.com/make/how-to-run-a-user-script-with-the-pijuice-power-events/
Thank you for building such a useful extra!

shawaj, 2020-10-24:
@lotte67890 closing this now as it seems you have everything working. Feel free to reply if you do not agree

closed 2020-10-24 by shawaj