PiSupply/PiJuice #550: power thru the raspi USB

pescadorj, 2020-09-28:


pescadorj, 2020-09-28:
When the power is supplied thru the raspi USB port the battery is not charged no blue led flashing. Is it possible that the supplied voltage is lower then the voltage supplied to the GPIO pins? What should the  the voltage be at a minimum?
Raspberry pi 3B pijuice firmware 1.4. Should work without a pogo pin, right?

tvoverbeek, 2020-09-28:
I guess he found the 'GPIO input enabled' setting

shawaj, 2020-10-24:
@pescadorj closing as it seems you resolved. If not please let us know

closed 2020-10-24 by shawaj

pescadorj, 2020-10-24:
More or less. If de power supply voltage on the raspi usb is lower then 5.2v the pijuice can not charge the battery, led is not flashing blue.Regards Jan VisserVerzonden vanaf Samsung-tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/PiSupply/PiJuice/issues/550#issuecomment-715644182",
"url": "https://github.com/PiSupply/PiJuice/issues/550#issuecomment-715644182",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]