PiSupply/PiJuice #551: Rtc synch between PiJuice Zero and RPiZW

FernandaRamosTL, 2020-09-30:
Hello! My order arrived today, and it was very fast considering the situation. I tried using that solution (sudo hwclock -s and the config txt including dtoverlay=i2c-rtc,ds1339) for making the time right when the internet is not present, the same way I did with my Pijuice Hat (the big one) units and the RPI 3b. This solution is not working wiht the pijuice zero + rpi zero. I wonder if the rtc module is different? Perhaps another different instruction is needed at the config.txt file when using this set of equipment (pijuice zero plus raspberry pi zero w)? Thank you!

tvoverbeek, 2020-09-30:
Should work. However the RPI zero is slower than the 3B.
When do you try to do the `sudo hwclock -s`? If e.g. from /etc/rc.local it might be too early (before i2c is initialized).
What is the output from `i2cdetect -y 1`?

FernandaRamosTL, 2020-09-30:
Hi! The output is correct, shows UU at 68.
Do you think that I should write a bash script where I tell it to sudo hwclock -s for when it reboots? Yes, the sudo hwclock -s is written on rclocal. 
I will not be able to try it now, the system is packed to be deployed, but I will have a twin system within 3 or 4 weeks to perform the tests.
Perhaps the problem is that rpizw is very very slow... let us wait and see :)

tvoverbeek, 2020-09-30:
You could use a cron script at @reboot.
At the start of the script you should check if I2C and the PiJuice is available, i.e by checking pijuice.status.GetStatus()
in a loop until a successful response. Especially at cold boot the PiJuice firmware also needs time to get started.
Alternatively use a long enough pause using sleep at the beginning of the script.

FernandaRamosTL, 2020-10-05:
Dear Ton, I am sorry for taking so long to give you an answer. I will try making the cron task and I guess that it will work, but only when I have the twin system with me.
I would have another question for you:
Is it possible for me to schedule the pij for turning the system on, in a minute period, only a few hours a day?
For instance:
Turn on every 5 minutes during the day time (from 6am to 6pm). Can I do it?
Thank you!

tvoverbeek, 2020-10-05:
Can be done, but you need the help from the Pi.
During the day you can use the recurring wakeup by setting the alarm to every 5 minutes every hour.
At the start of your job check the time and when it is >= end time (6pm) then reprogram the alarm to  wakeup at 6am the next day.
At the first job of the day (at 6am) reprogram the alarm again to every 5 minutes every hour.

FernandaRamosTL, 2020-10-05:
I see, so it cannot be automated... no problem. Do you think is there some kind of electronic or hardware related issue if the RPI is booted so many times a day during long long periods (months or years)?

shawaj, 2020-12-05:
@FernandaRamosTL do you need any further assistance at all with this? Or can we close the issue?