PiSupply/PiJuice #552: Clarification on IoCurrent and BatteryCurrent

mndtrythnkng, 2020-09-30:
Hello, Python / rPi newbie here, trying to get my head around the API calls as I try to set up an off-grid server powered by solar. I'm using an rPi 4 with a PiJuice HAT and 1820mAh battery.

I'm successfully making calls to the API to return various stats including the current, using:

`pijuice.status.GetIoCurrent()["data"]`
`pijuice.status.GetBatteryCurrent()["data"]`

I've read the documentation, but I'm still unclear as to what these two results are actually showing. When PiJuice is connected to a power source it returns the following values:

```
IoCurrent: 2778
BatteryCurrent: -1376
```

If the power supply is interrupted (i.e. running from battery) it returns these values:

```
IoCurrent: 2778
BatteryCurrent: 570
```

Is `GetIoCurrent` returning the current passing from the power source in order to keep the PiJuice battery charged, or is it returning the power currently being used by the rPi? And does `BatteryCurrent` only return a positive value if the rPi is being powered from the PiJuice battery?

Any help/clarification would be appreciated!

tvoverbeek, 2020-09-30:
IoCurrent is the current to/from the 5V GPIO pins.
Positive IoCurrent means current from PiJuice to the RPi.
BatteryCurrent positive means current from battery to PiJuice/RPi
Battery current negative means current to battery (i.e. charge current).
If you would power via the RPi you would see negative IoCurrent.
Values are mA.
Note: There are no direct current measurements. They are estimated by the firmware.

mndtrythnkng, 2020-09-30:
Thanks @tvoverbeek, that's really clear.

Is is safe to use the IoCurrent reading as an approximation of the rPi's power consumption? i.e. in its current configuration it's requiring 2778mAh to continue running?

tvoverbeek, 2020-09-30:
It is probably a good approximation. 2.7A means you have some peripherals also drawing current
(TFT screen, USB disk?).
2.7A is at the limit of what the PiJuice can supply when on battery.
My Pi4B (headless, running desktop and VNC server) uses around 900mA according GetIoCurrent().

mndtrythnkng, 2020-09-30:
Fantastic, thank you.
Yes, I currently have a monitor and keyboard/mouse attached while I familiarise myself with everything. (Thonny's built-in debug assistant is proving very useful!) Aim eventually to go headless, possibly even switching to a PiZero for even lower power consumption.

closed 2020-09-30 by mndtrythnkng