PiSupply/PiJuice #554: Control PiJuice from Arduino, connecting the J5 only if the battery level if high enough.

Passiday, 2020-10-03:
Does PiJuice HAT read and write i2c also before the outgoing power is activated? In my project I plan to use Arduino to talk to the PiJuice, use it as a sort of gatekeeper that turns on a red led if the battery level is under certain threshold and connects the J5 if the battery level is ok. I know I could put an Arduino-controlled mosfet to control if the power reaches Pi, but I just want to know if I have to go that way.

tvoverbeek, 2020-10-03:
As long as the PiJuice has power (i.e. the on-board micro is running) it will respond to I2C communications.
Note you do not need an Arduino to do the gatekeeping.
Use the low battery and wakeup-on-charge events to achieve the same thing.
On low battery level shut down the Pi and when the charging reaches the wakeup-on charge level it will turn on the Pi.

Passiday, 2020-10-03:
Awesome. Is this true also for the low power deep sleep state mentioned in the readme? I am still undecided about the architecture of the final hardware config. The project is a toy that is mainly driven by Raspberry Pi Zero, it has 10 RGB LEDs and a speaker. It should automatically shut down when not used for about 10 minutes, and can be started with push of a button. But I want it to have some LED animation while the Pi is booting up and shutting down, so there's Arduino to drive that. And I thought it would be nice to flash the LEDs red, if user tries to power it up when the battery level is low. So this is why I think I need the Arduino to do the gatekeeping - the PiJuice would power Arduino, but it would be Arduino who decides should the power be passed further to the Pi. Well, I see now that there would be noone to talk to the PiJuice, when there's no power at it's output. But it's still very cool that it is responding at that state, though. I'm very impressed by the feature set, it's a job done very well.

closed 2020-10-04 by Passiday