PiSupply/PiJuice #555: Remove of power does not trigger System Event (no power)

morningshane, 2020-10-05:
Trying to test the below script, it did not trigger anything after I remove the power from the PiJuice. There is also no power  connected to the Pi Zero W. I have tried other system events like SYS_FUNC_REBOOT, none of them are working. The LED works normal (solid green when no power, flash when charging) I'm using python 2.7

#!/usr/bin/python3

from pijuice import PiJuice
import os
pijuice = PiJuice(1, 0x14)

# Remove power to PiJuice MCU IO pins
pijuice.power.SetSystemPowerSwitch(0)

# Set wakeup
pijuice.power.SetWakeUpOnCharge(0.0)

# Remove 5V power to RPi after 60 seconds
pijuice.power.SetPowerOff(60)

# Shut down the RPi
os.system("sudo halt")

[image]
[image]
[image]

ChristopherRush, 2020-10-05:
Check that the owner of your python script is user 'pi' and that the script can be executed. It is documented here - https://github.com/PiSupply/PiJuice/tree/master/Software#user-scripts-menu

Are you able to run the script from the command line?

morningshane, 2020-10-05:
> Check that the owner of your python script is user 'pi' and that the script can be executed. It is documented here - https://github.com/PiSupply/PiJuice/tree/master/Software#user-scripts-menu
> 
> Are you able to run the script from the command line?

thanks for the reply
[image]
It shows "pi" is the user

sudo python shutdown.py is working, the Pi will shutdown
sudo chmod +x shutdown.py, nothing happend.
[image]

tvoverbeek, 2020-10-05:
With `sudo chmod +x shutdown.py` you tried to add the execute (x) permission to the file.
According to your first screenshot shutdown.oy already had th 'x' permission, so of course nothing happened.
Can you run shutdown.py directly (i.e `/home/pi/shutdown.py` (without python in front)?
Can you list the contents of your /home/pi/shutdown.py?

morningshane, 2020-10-06:
> With `sudo chmod +x shutdown.py` you tried to add the execute (x) permission to the file.
> According to your first screenshot shutdown.oy already had th 'x' permission, so of course nothing happened. [...]

I found the issue was system task enable is not checked..... and it is working now,
This is what I want to do

- everytime the pi is wake up, it will run a script from cron @reboot. 
- Then the pijuice needs to shutdown the Pi
- Pijuice enter a low powered state until next wakeup
- Repeat the above steps...

[image]

and here is the Pijuice settings
[image]
[image]
[image]
[image]

tvoverbeek, 2020-10-06:
You wrote:
- everytime the pi is wake up, it will run a script from cron @reboot.
- Then the pijuice needs to shutdown the Pi            <=====
- Pijuice enter a low powered state until next wakeup
- Repeat the above steps...

The PiJuice cannot shut down the Pi.
The shutdown has to be commanded by the Pi (as in your script).
Your setup should work to wake the Pi up every 3 minutes.
OK to close the issue?

morningshane, 2020-10-21:
> You wrote:
>
> * everytime the pi is wake up, it will run a script from cron @reboot.
> * Then the pijuice needs to shutdown the Pi            <=====
> * Pijuice enter a low powered state until next wakeup [...]

Sorry for replying this late, I was on a holiday for a while and just saw the reply.
The wakeup is not working.....
I tried SYS_FUNC_HALT_POW_OFF for No Power instead of using my own script,  after removing the usb power source, the pi will not wake up too.
but when I connect the usb power source to the Pijuice, the Pi will wake up.

So it's not the issue of my script and I guess it's the issue of triggering the wakeup.
[image]
[image]
[image]
[image]

morningshane, 2020-10-21:
Not even a single successful wake up so far......

tvoverbeek, 2020-10-21:
Now I am confused.
Are you talking about wakeup when external power is reapplied or about the periodic 3 minutes wakeup?
For the 3 minute period wakeup there could be problems if you have an internet connection.

morningshane, 2020-10-21:
> Now I am confused.
> Are you talking about wakeup when external power is reapplied or about the periodic 3 minutes wakeup?
> For the 3 minute period wakeup there could be problems if you have an internet connection.

I'm talking about 3 minute period wakeup. Yes I do have an internet connection so what should I do?

thanks

tvoverbeek, 2020-10-21:
OK. Are we talking about the PiJuice Zero or the PiJuice HAT?
Do you have a dtoverlay line for the rtc in /boot/config.txt?

morningshane, 2020-10-21:
> OK. Are we talking about the PiJuice Zero or the PiJuice HAT?
> Do you have a dtoverlay line for the rtc in /boot/config.txt?

Just the Pi Zero, I have tried to change the EEPROM address to 52 but still doesn't work. Do I need to add dtoverlay=i2c-rtc,ds1307 to /boot/config.txt?

tvoverbeek, 2020-10-21:
Not the Pi Zero W, but the PiJuice Zero with external battery?
Anyway, the PiJuice Zero does not have a HAT EEPROM, hence changing the EEPROM address to 0x52 has no effect for the Pijuice Zero.
What is the output from `i2cdetect -y 1`?

morningshane, 2020-10-21:
> Not the Pi Zero W, but the PiJuice Zero with external battery?
> Anyway, the PiJuice Zero does not have a HAT EEPROM, hence changing the EEPROM address to 0x52 has no effect for the Pijuice Zero.
> What is the output from `i2cdetect -y 1`?

I'm using the PiJuice not the PiJuice Zero because I need to use the GPIO pins to connect to my E-paper display. here is the output
[image]

morningshane, 2020-10-22:
> Not the Pi Zero W, but the PiJuice Zero with external battery?
> Anyway, the PiJuice Zero does not have a HAT EEPROM, hence changing the EEPROM address to 0x52 has no effect for the Pijuice Zero.
> What is the output from `i2cdetect -y 1`?

Changed my Pi Zero W to my newly purchased Pi 4, no luck

tvoverbeek, 2020-10-22:
Do you still get 68 instead of 'UU" for the RTC in the i2cdetect output on the Pi 4?

morningshane, 2020-10-25:
> Do you still get 68 instead of 'UU" for the RTC in the i2cdetect output on the Pi 4?

I got 'UU' this time. but the Pi is still not waking up
[image]

tvoverbeek, 2020-10-25:
Since your Pi has an internet connection, the systemd-timesyncd daemon will synchronize both the system time and the RTC with internet time. Since the OS assumes it has control of the RTC (the 'UU' in the i2cdetect output) it will also clear the alarm in the RTC. Hence no repeated wakeup.
See also https://github.com/PiSupply/PiJuice/tree/master/Software#repeated-wakeup
However the script there assumes no internet connection.
You need to do the `pj.rtcAlarm.SetWakeupEnabled(True)` just before you shut down the Pi
You can also check if it is enabled with `pj.rtcAlarm.GetControlStatus()`.

morningshane, 2020-10-26:
> Since your Pi has an internet connection, the systemd-timesyncd daemon will synchronize both the system time and the RTC with internet time. Since the OS assumes it has control of the RTC (the 'UU' [...]

I have already put pj.rtcAlarm.SetWakeupEnabled(True) which you could see from our previous conversation. it didn't work....

no matter what I tried, the Pi is not wake up. Not just repeated wakeup, even I select '2 min' wakeup instead of 'every 2 mins'. Just no wakeup at all.

[image]
[image]

morningshane, 2020-10-26:
> Since your Pi has an internet connection, the systemd-timesyncd daemon will synchronize both the system time and the RTC with internet time. Since the OS assumes it has control of the RTC (the 'UU' [...]

I recorded 2 short videos here so you can see the setup and what happened after I unplugged the power.
https://youtu.be/Gb9MhI0MJPQ
https://youtu.be/7vABi4MmXJ8

shawaj, 2020-10-26:
The way you have it set is to wake up at 2 minutes past midnight. Try ticking the every day and every hour options then it will wake up at 2 minutes past every hour

morningshane, 2020-10-26:
> The way you have it set is to wake up at 2 minutes past midnight. Try ticking the every day and every hour options then it will wake up at 2 minutes past every hour

I see.But the issue is no matter I select 2 mins or every 2 mins, it will not wake up the Pi.

shawaj, 2020-10-26:
Have you tried what I said? You'd have to wait until 2 minutes past the hour to test it.

tvoverbeek, 2020-10-26:
As @shawaj said, the alarm setting is for the RTC clock time, not an elapsed time from powerdown.
When you choose day=0 (as in your settings) the Set Alarm does not give you an error message (It should), but
the alarm will not be set.
If you want to wakeup 2 minutes after power down, you will have to get the current time form the RTC (in UTC) and set the alarm 2 minutes later. The easiest way is to select 'Every hour", 'Every day'.
Set the minutes to the current minute + 2 (modulo 60) and keep the seconds from the current time.
This has to be added to your shutdown script before you se the wakeup enabled.

morningshane, 2020-10-30:
> Have you tried what I said? You'd have to wait until 2 minutes past the hour to test it.

> As @shawaj said, the alarm setting is for the RTC clock time, not an elapsed time from powerdown.
> When you choose day=0 (as in your settings) the Set Alarm does not give you an error message (It should), but
> the alarm will not be set.
> If you want to wakeup 2 minutes after power down, you will have to get the current time from the RTC (in UTC) and set the alarm 2 minutes later. The easiest way is to select 'Every hour", 'Every day'.
> Set the minutes to the current minute + 2 (modulo 60) and keep the seconds from the current time.
> This has to be added to your shutdown script before you se the wakeup enabled.

Thanks very much for both of you, now the repeated wakeup is working!!!

Just one last question here that may need your help.

I'm pretty new to this field, so what I want to do is to wake up the pi very 2 hours and run a script located at "/home/pi/infowindow.py" but the script needs to be ran**when internet is on**.  then finally shutdown the pi and make the piJuice enter into a low power consumption and wait for the next wakeup. 

what exactly do I need to do?

thanks in advance.

shawaj, 2020-10-30:
@morningshane you could add a test to the script that checks for internet. Quit the script if no internet, continue if successful.

Something like this:
```
loop_value = True
while loop_value:
        try:
                urlopen("http://google.com")
```