PiSupply/PiJuice #556: pijuice zero w gpio header extender

bpehrson, 2020-10-10:
Any advice on how to connect external sensors to the gpio header on a pijuice/rpi zero w combination? I looked in vain for some sort of extender. Do I have to solder on the downside of the rpi zero w?

ChristopherRush, 2020-10-10:
You could use a pHAT or HAT expansion board or if you only need to stack 1
more board and extending header with longer male pins. If I remember
correctly the pins need to be about 11mm.

bpehrson, 2020-10-10:
I have seen those for the normal rpi form factor but not for the pi zero, and I have spent a lot of time on this. If you know that they exist, could you provide a pointer! Thanks a lot for responding!

ChristopherRush, 2020-10-10:
Are you looking for an expansion board or extending header?

bpehrson, 2020-10-10:
My acute problem is to connect a few dupont contacted wires from external sensors to some gpio pins, a coouple of  i2c-gadhets , simple pulse counters and relay controls. The wires have so far been connected directly to the RPi zero headers buth the full length of this header was of course eaten up by the pijuice zero card. 
A 2*20 header with enough longer pins would work but I have not found any.
How would an expansion board be connected and would that solve my problem?

shawaj, 2020-10-10:
@bpehrson an expansion board such as this one would allow you to connect multiple things at once - https://uk.pi-supply.com/products/nebra-ip67-case-gateway-hat-mounting-and-expansion-board

Or your could use a header like this - https://shop.pimoroni.com/products/male-40-pin-2x20-hat-header?variant=15387338113107

Or just a standard header soldered to the Pi and a stacking female header on top like this - https://shop.4tronix.co.uk/products/custom-2x20

shawaj, 2020-10-24:
Closing this issue now as think it is resolved. Feel free to reply back if you don't agree

closed 2020-10-24 by shawaj