PiSupply/PiJuice #557: Upgrade from version 1.3 to 1.4 wrong battery profile

DazFahy, 2020-10-14:
Hi, I am currently converting the Python code to C# as a module under the GitHub [Microsoft dotnet iot devices project].(https://github.com/dotnet/iot/tree/master/src/devices) 

Therefore I do not have the Python tools installed on my Raspberry Pi 4. I have recently upgraded my PiJuice from version 1.3 to version 1.4, after doing this I have noticed that the battery profile is still being returned as zero which under version 1.3 was the BP6X_1400 battery profile. However now under version 1.4 this is the PJZERO_1000 battery profile, is there anything you have to do after the upgrade to version 1.4 to get the correct battery profile?

The code I am referring to regarding the battery profiles is

```
if fwver >= 0x14:
    self.batteryProfiles = ['PJZERO_1000', 'BP7X_1820', 'SNN5843_2300', 'PJLIPO_12000', 'PJLIPO_5000', 'PJBP7X_1600', 'PJSNN5843_1300', 'PJZERO_1200', 'BP6X_1400', 'PJLIPO_600', 'PJLIPO_500']
elif fwver == 0x13:
    self.batteryProfiles = ['BP6X_1400', 'BP7X_1820', 'SNN5843_2300', 'PJLIPO_12000', 'PJLIPO_5000', 'PJBP7X_1600', 'PJSNN5843_1300', 'PJZERO_1200', 'PJZERO_1000', 'PJLIPO_600', 'PJLIPO_500']
else:
    self.batteryProfiles = ['BP6X', 'BP7X', 'SNN5843', 'LIPO8047109']```

tvoverbeek, 2020-10-14:
Yes, the order of the battery profiles has changed. This is to get a safe default for the PiJuice Zero which does not have dip switches.
The PiJuice HAT can select one of the first four profiles by dip switch. The profile id internally is then set to 0xFF (DEFAULT setting). If the user has selected a profile the profile source is set to HOST and the id stored in non-volatile memory.
For the nitty gritty see the various battery profile functions in pijuice.py.
You would have to replicate the logic in C#.

DazFahy, 2020-10-14:
Thanks for the information. I've got the standard battery which is the BP7X_1820 so even under 1.3 it was returning me zero which is the BP6X_1400. I will check the dip switches to make sure they have not changed.

DazFahy, 2020-10-14:
Was indeed the dip switches, I've now changed these to match the BP7X_1820.

closed 2020-10-14 by DazFahy