PiSupply/PiJuice #558: Power on PiJuice/Pi when power cable is attached

davidmtod, 2020-10-14:
Hi,

I am trying to make my PiJuice/Pi combo start up when I connect a power cable.  This is in case my solar panel solution isn’t powerful enough to charge the battery and I need to charge the battery / run the solution from the mains.  I toggled the trigger to wake on charge level and set the charge threshold to 10%.  I then did a sudo halt, unplugged the adapter then plugged it back in.  The led cane on the PiJuice but the Pi did not start.  Could someone advise what I’ve done wrong and how you achieve this?

Thanks,

David 

[image]

tvoverbeek, 2020-10-14:
I am guessing you commanded the 'sudo halt' yourself or by a user program.
In that case you have to set the wakeup-enabled flag yourself before you shutdown the Pi.
Use the pijuice.power.SetWakeUpEnabled(10) function.
The triggerlevel on the System Task page is set by the pijuice service before shutdown when you use on of the SYSTEM_HALT_xxx functions.
See also https://learn.pi-supply.com/make/how-to-run-a-user-script-with-the-pijuice-power-events/

davidmtod, 2020-10-14:
> I am guessing you commanded the 'sudo halt' yourself or by a user program.
> In that case you have to set the wakeup-enabled flag yourself before you shutdown the Pi.
> Use the [...]

Hi tvoverbeek,

Thank you for your reply.  You are correct, I was manually issuing a sudo halt.  I followed the instructions in the link you gave me and it appears to do exactly what I need it to do.  One question:  if the battery drains completely and the PiJuice isn’t able to retain its low power state I take it I’ll need to start it with the SW1 power switch after plugging in a power adaptor, is that correct?

Thanks,

David

tvoverbeek, 2020-10-14:
Yes, The wakeuponcharge is not saved in non-volatile memory.
Hence when the MCU loses power it starts from default values.
Try to enable 'No battery turn on' in the HAT config and see if that allows you to start the Pi when the battery is drained.
The 'No battery turn on' setting is saved in non-volatile memory.

davidmtod, 2020-10-14:
> Yes, The wakeuponcharge is not saved in non-volatile memory.
> Hence when the MCU loses power it starts from default values.
> Try to enable 'No battery turn on' in the HAT config and see if that allows [...]

Thanks for the suggestion.  I tried enabling this setting but it didn’t make any difference.  It might be as the battery wasn’t actually dead.  I’ll try disconnecting the battery and test again.

tvoverbeek, 2020-10-15:
As long as the battery still supplies power to the PiJuice MCU (-> Firmware still running even in low power mode) the
'no battery turn on' has no effect since it still detects the battery.

davidmtod, 2020-10-16:
> As long as the battery still supplies power to the PiJuice MCU (-> Firmware still running even in low power mode) the
> 'no battery turn on' has no effect since it still detects the battery.

Thanks for your reply.  I inadvertently tested this overnight when I left my pijuice/pi turned on thinking it was powered from the mains when the wall switch wasn't down and the battery died overnight.  When I plugged it in this morning with the battery at 0% it fired up as you indicated.

Thanks for your help with this.

David

closed 2020-10-16 by davidmtod