PiSupply/PiJuice #559: J5 switch not powering on RPi

davidmtod, 2020-10-14:
Hi,

Am I correct in saying the J5 is exactly the same function as SW1?  I connected a switch to J5 but it doesn’t power on the RPi from cold (sudo halt) whereas the SW1 micro switch does power on the RPi. If the J5 functionality is the same as SW1 then I think I’m doing something wrong.   Is there anything that needs enabled on the HAT config to allow using J5 for an external power supply?

Cheers,

David

ChristopherRush, 2020-10-14:
What kind of switch are you using?

davidmtod, 2020-10-14:
> What kind of switch are you using?
> […](#)

I’m using this switch:

Coolais 12mm Mini Push Button Switch Momentary Waterproof Metal Push Button Micro Switches SPST NO Momentary Switch 3A 250V 5 Sets/Pack PBSM-02 (High Head) https://www.amazon.co.uk/dp/B07L4LSXNR/ref=cm_sw_r_cp_api_i_T42HFbDB7CPPJ

It may be this switch isn’t compatible, I’m not sure.  Or my soldering is at fault!

ChristopherRush, 2020-10-16:
Switch looks ok to use in this application. If you have a multimeter it could be worth just testing the functionality of the switch using the continuity test function on the meter. From a software point of view there isn't anything that needs to be configured for use with the J5 header as it is electrical identical to SW1.

davidmtod, 2020-10-16:
> Switch looks ok to use in this application. If you have a multimeter it could be worth just testing the functionality of the switch using the continuity test function on the meter. From a software [...]

Thanks Christopher,

The behaviour I'm seeing now is that the button doesn't start the pi, has no effect during normal use of the pi but a single button press shuts the pi down. So the button is having some effect just not the desired one.

I tested it with a multimeter.  Continuity when I press the button, open when not pressed.

I think I'm going to give up before I fry the circuit board though.  I've done a lot of soldering / desoldering J5.  Can I just check:. J5 has two holes, I take it I just solder each wire of the button to each of the holes? Does it matter what way round?

Thanks,

David

davidmtod, 2020-10-16:
Hi Christopher,

I managed to sort it by going into pijuice_cli and setting the sw2 button functionality to be the same as sw1.  This is now working with a single press turning the pi on and prolonged press halting the pi.  I followed the guidelines here:

https://github.com/PiSupply/PiJuice/issues/467

Thanks for your help with this.

Cheers,

David

closed 2020-10-16 by davidmtod

shawaj, 2020-10-16:
Sorry, didn't realise you were using a PiJuice Zero! Apologies 👍

davidmtod, 2020-10-17:
> Sorry, didn't realise you were using a PiJuice Zero! Apologies 👍

That was my fault - I didn't state it was a pijuice zero as I thought the only difference between them was form factor.  Every day's a school day 🙂 still, a great product nonetheless!

shawaj, 2020-10-17:
Just for anyone else for future reference, the info on the PiJuice Zero buttons is here https://github.com/PiSupply/PiJuice/tree/master/Hardware#zero

And the HAT just above :-)