PiSupply/PiJuice #563: RTC detection not stable on Pi Zero + PiJuice Zero

Passiday, 2020-10-19:
I have Pi Zero with PiJuice Zero. Ther's limited space in the box, the devices are squeezed there somwehat tight, so the PiJuice is connected to the Pi Zero via 6-wire cable that connects just the top six pins. I have consulted with tech support in Discord that it's ok, provided that I manually enable the RTC overlay.  

So I have added `dtoverlay=i2c-rtc,ds1339` to the `/boot/config.txt` and rebooted.

At first, the output of `sudo hwclock -r` was working, and the `i2cdetect -y 1` showed `UU` at address 0x68. But, for some reason, time to time something happens and RTC is not recognized (`Cannot access the Hardware Clock via any known method.` error returned when executing `sudo hwclock -r`), and `i2cdetect -y 1` shows `68` at address 0x68.

There are no other HATs installed, `/proc/device-tree/hat` does not exist. The I2C communication with PiJuice is normal, ie I can read the state in Python.

Any ideas how to get the RTC reliably present at each boot?

tvoverbeek, 2020-10-20:
The PiJuice Zerp does not have an ID-EEPROM like full size HATs.
Hence no automatic load of the RTC device driver (on the HAT this is done due to the device-tree blob in the ID-EEPROM).
Also that is why /proc/device-tree/hat does not exist.
The RTC is not a separate hardware RTC, but simulated by the PiJuice firmware. The firmware has to be active (finished its startup) before the RTC will be recognized.
Before you try to use `sudo hwclock -r` first try to access the PiJuice by using the pijuice.status.GetStatus() comand in a loop until you get a correct reply (error = 'NO_ERROR')

Passiday, 2020-10-20:
Today I disconnected the battery from PiJuice, but left the controller board connected to Pi. Then connected USB directly to Pi Zero, booted, and the hwclock was there, albeit reset to 2000-01-01. I could set the date+time and read it correctly. The shut down, disconnected the USB from Pi, connected the battery. Booted, no hwclock. Not sure if all these acrobatics are somehow connected with the unstable RTC loading, but at least I have now this state when RTC is not recognized but the PiJuice interface seems to be working. Here's excerpt from my terminal session:

    pi@PiZero:~ $ python3
    Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
    [GCC 8.2.0] on linux
    Type "help", "copyright", "credits" or "license" for more information.
    >>> from pijuice import PiJuice
    >>> pijuice = PiJuice(1, 0x14)
    >>> print(pijuice.status.GetStatus())
    {'data': {'isFault': False, 'isButton': False, 'battery': 'NORMAL', 'powerInput': 'NOT_PRESENT', 'powerInput5vIo': 'NOT_PRESENT'}, 'error': 'NO_ERROR'}
    >>> print(pijuice.status.GetChargeLevel())
    {'data': 60, 'error': 'NO_ERROR'}
    >>> exit()
    pi@PiZero:~ $ sudo hwclock -r
    hwclock: Cannot access the Hardware Clock via any known method.
    hwclock: Use the --verbose option to see the details of our search for an access method.
    pi@PiZero:~ $ date
    O  okt 20 19:05:31 EEST 2020
    pi@PiZero:~ $ i2cdetect -y 1
         0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
    00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
    10: -- -- -- -- 14 -- -- -- -- -- -- -- -- -- -- -- 
    20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
    60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
    70: -- -- -- -- -- -- -- --                         
    pi@PiZero:~ $ 

Although there's NO_ERROR from PiJuice Python interface, the hwclock is still not recognized.

clucani, 2020-11-02:
Hey, yep I'm having exactly the same issue. Any help much appreciated.

Passiday, 2020-11-02:
Don't know what to think of it, but I started afresh from the latest Raspberry Pi OS image, on a clean SD card, and the problem just magically disappeared.

clucani, 2020-11-03:
Oh that is interesting. I'll give that a go as well. Thanks

shawaj, 2020-11-12:
@Passiday @clucani are you all sorted now?

clucani, 2020-11-12:
@shawaj - I haven't had a chance to test it out yet, when I do I'll post back.

shawaj, 2020-11-12:
Thane

gobees, 2023-12-01:
I don't know if OP resolved their problem but I have exactly the same problem.  I started with fresh raspbian update on brand new SD card followed by install of pijuice.  I believe the OP's problem still exists and I'm hoping for another look at the issue.

Also, see my post [here](https://github.com/PiSupply/PiJuice/discussions/1041)

[image]