PiSupply/PiJuice #564: PiJuice Zero not starting Pi Zero when motion detected

davidmtod, 2020-10-20:
I'm trying to install and configure the NatureBytes project.  The Pi shuts down correctly after 10 minutes of non motion via the service/script. One of the issues I am encountering is that the Pi wont restart when there is motion in front of the sensor.  I have tested the sensor when the Pi is on and it detects  motion correctly. I followed the guide to configure IO2 in the PiJuice_Cli.

Can anyone suggest anything else I need to try?

It's a Pi Zero W / PiJuice Zero combination I am using.  The pin layout I am using for the PIR is:
VCC Pin 2 (5V)
DOUT Pin 11 (GPIO 17)
Ground Pin 14

Thanks,

David

davidmtod, 2020-10-21:
Hi,

I've been looking into this further, reading documentation etc and I have seen reference to J3 on the PiJuice Zero (see attached pic).  J3 has an IO2 slot as well as ground and 3.3v slots.  In order to get the PiJuice Zero to start my Pi when motion is detected do I need to use these pins to connect the PIR instead of the GPIO pins that I'm currently using?  Is there any IO2 functionality on the GPIO (apologies if that's a daft question - i'm still trying to find my feet with the electronics).

[image]

Also what does IO1 do?  I've seen that in the pijuice_cli under IO.

Thanks,

David

shawaj, 2020-10-21:
We have a guide here https://learn.pi-supply.com/make/pijuice-naturejuice-project/

davidmtod, 2020-10-21:
Hi Aaron,

Thanks for your reply and sending this on.  I take it the only way to achieve this functionality is through soldering the PIR wires onto the board instead of using the GPIO?  The guide talks about modifying the PIR to use 3.3v.  Could I not just use the 5v pin on the board so I don’t have to modify the PIR?

Thanks,

David

shawaj, 2020-10-21:
It's recommended to follow the guide exactly as that's the only way we were able to get it to work.

davidmtod, 2020-10-21:
Thanks Aaron,

But why go to the bother of modifying the PIR to usev3.3v instead of 5 and to use a 3.3v input when you could just use the 5v input as-is?

Thanks,

David

tvoverbeek, 2020-10-21:
The PIR has to wakeup the Pi when motion is detected.
When the Pi shuts down the 5v to the Pi from the PiJuice is switched off. This is the same 5V going to the PIR.
So no way to wake up the Pi when the PIR is fed from the 5V.
The 3V3 stays on because the PiJuice MCU has to remain running, even when in low power mode.

davidmtod, 2020-10-21:
I didn't realise that, thank you for the explanation.  I'll probably abandon this plan as I'm not comfortable with modifying the PIR.

Thanks again for your help.

David

tvoverbeek, 2020-10-21:
@davidmtod No need to abandon your project. Have you seen this https://www.adafruit.com/product/4667 or https://www.adafruit.com/product/4666? A mini PIR sensor on 3.3V. For a naturebytes project these should work.

davidmtod, 2020-10-21:
I hadn't seen them, thanks for the links.  I'll give it another shot :)

davidmtod, 2020-11-10:
I have ordered the 3.3v PIR and I’m waiting for it to arrive to test it.  In the mean time I’ve soldered the header on to J3.  I came across the following article which suggested I could use my current PIR by connecting the VCC pin to the header in the top right corner in order to power from 3.3v.  Is this accurate?  I am currently testing it but It’s not switching the pi on with motion.  Sometimes it starts up if I wiggles the wires on the PiJuice header.  This could be down to my bad soldering of the header or something else as the header seems solid.  In your opinion is this route worth pursuing? I.e. is this pin a 3.3v input?

https://randomnerdtutorials.com/modifying-cheap-pir-motion-sensor-to-work-at-3-3v/

[image]

Thanks,

David

davidmtod, 2020-11-11:
I seem to have got it working consistently now with the above set up.  I redid my soldering and it's a lot neater now.

closed 2020-11-11 by davidmtod

shawaj, 2020-11-11:
@davidmtod awesome, nice work. And thanks for letting us know!