PiSupply/PiJuice #565: pijuice_cli crash

blanco-ether, 2020-10-20:
crashes when I select General.  Firmware is 1.4.  I assume pijuice_cli is same vintage.

Have 5 RPi's with PiJuice.  They all do it.

Would configuration help you?

Traceback (most recent call last):
  File "/usr/bin/pijuice_cli.py", line 2007, in <module>
    loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 286, in run
    self._run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 384, in _run
    self.event_loop.run()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 788, in run
    self._loop()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 816, in _loop
    self._entering_idle()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 777, in _entering_idle
    callback()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 572, in entering_idle
    self.draw_screen()
  File "/usr/lib/python3/dist-packages/urwid/main_loop.py", line 586, in draw_screen
    canvas = self._topmost_widget.render(self.screen_size, focus=True)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 144, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 783, in render
    self.top_w_size(size, left, right, top, bottom), focus)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 144, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 1765, in render
    canv = get_delegate(self).render(size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 144, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 1529, in render
    canv = w.render((maxcol, rows), focus=focus and item_focus)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 144, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/container.py", line 2087, in render
    focus = focus and self.focus_position == i)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 144, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 567, in render
    canv = self._original_widget.render((maxcol,)+size[1:], focus)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 144, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/decoration.py", line 567, in render
    canv = self._original_widget.render((maxcol,)+size[1:], focus)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 144, in cached_render
    canv = fn(self, size, focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 471, in render
    (maxcol, maxrow), focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 353, in calculate_visible
    self._set_focus_complete( (maxcol, maxrow), focus )
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 718, in _set_focus_complete
    (maxcol,maxrow), focus)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 688, in _set_focus_first_selectable
    (maxcol, maxrow), focus=focus)
  File "/usr/lib/python3/dist-packages/urwid/listbox.py", line 420, in calculate_visible
    n_rows = next.rows( (maxcol,) )
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 1775, in <lambda>
    rows = property(lambda self:get_delegate(self).rows)
  File "/usr/lib/python3/dist-packages/urwid/widget.py", line 1775, in <lambda>
    rows = property(lambda self:get_delegate(self).rows)
AttributeError: 'NoneType' object has no attribute 'rows'

tvoverbeek, 2020-10-21:
What is the version of the installed pijuice-base package?
Use `apt policy pijuice-base`.
My output is:
```
$ apt policy pijuice-base
pijuice-base:
  Installed: 1.6
  Candidate: 1.6
  Version table:
 *** 1.6 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
```

blanco-ether, 2020-10-23:
Mine shows

pijuice-base:
  Installed: 1.5
  Candidate: 1.5
  Version table:
 *** 1.5 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status

I updated pijuice-gui and the problem has gone away.

Sorry, I should have thought of this myself.

Thanks

shawaj, 2020-10-23:
Thanks for the update. Closing the issue now.

If you have any other problems feel free to start a new one :-)

closed 2020-10-23 by shawaj