PiSupply/PiJuice #566: Update where to buy list

devicenull, 2020-10-22:
A bunch of the links for US-based vendors in the where to buy list are dead.  For example, Newark no longer offers it for sale, and a bunch of the other vendors list 'coming soon'.

shawaj, 2020-10-22:
@devicenull thanks very much.

@tvoverbeek @ryanteck @ChristopherRush  maybe best to remove them altogether what do you think?

ChristopherRush, 2020-10-22:
I think we can update it based on the distrubters list on our website. We
don't necessarily need to link directly to the product page itself.

tvoverbeek, 2020-10-22:
@devicenull means the distribution list on the main github page which has not been updated for 2 years.
Also not all distributors on https://uk.pi-supply.com/pages/distributors are selling PiJuice and PiJuice related products.
I suppose you (=Pi Supply) know the distributors to whom you shipped the PiJuice, 
So either update it or delete it altogether.
My 5 cents.

shawaj, 2020-10-22:
Possibly easier to just Google at this point. We made the list originally to help people out when we had just started shipping. Probably served it's purpose now.

Will see what @JohnWhiting thinks and update. Will leave this issue open for now

commit 09f97a5b (2020-11-03): update where to buy section

closes #566
closed 2020-11-03 by shawaj