PiSupply/PiJuice #567: Updated battery suggestions?

devicenull, 2020-10-22:
Do you have any suggestions for batteries that are actually easy to obtain?  It's difficult to find BP7X these days, and ones you can find have date codes from 10 years ago.

I see that PiSupply still sells batteries, but I'd pay nearly as much in shipping as I would for the battery (I'm in the US).

Is a single 18650 cell a good option?  The voltage and capacity seem pretty similar, and they are readily available worldwide.

shawaj, 2020-10-22:
Which battery is it that you're looking for?

Is it the 1820mAh replacement battery? Or one of the larger ones?

devicenull, 2020-10-22:
I'd take either at this point.  I only need a few minutes of battery life.

I found an 18650 packaged with protection + a thermistor that I'm leaning towards at the moment: https://www.batteryspace.com/custom-li-ion-18650-battery-3-6v-2600mah-9-36wh-with-10k-thermistor---un38-3-passed.aspx?variation=11239

shawaj, 2020-10-22:
Digi-key have them

https://www.digikey.co.uk/products/en?keywords=Pis-1129

https://www.digikey.co.uk/products/en?keywords=Pis-1131

https://www.digikey.co.uk/product-detail/en/pi-supply/PIS-1189/1910-PIS-1189-ND/13283434

We sent them some stock the other day so should have them in stock for sale soon

shawaj, 2020-10-22:
Having said that, any single cell lipo will work. So the one you linked looks ok.

devicenull, 2020-10-22:
Ah, thanks!

It turns out... I had charging disabled.... not a dead battery.  I have no idea how it got disabled, but I didn't realize that had happened until I was reading some other issues here.  It would be nice if the status page displayed that!

shawaj, 2020-10-24:
Great glad to hear it's working. Will close this issue now ☺️

closed 2020-10-24 by shawaj