PiSupply/PiJuice #568: intelligent on/off switch

mrglobetrotter0088, 2020-10-23:
In the README file I read something about an intelligent on/off switch, which allows to have control on when the Raspberry Pi will be switched on or off without having to plug or unplug the PSU as you would normally have to do. I'm interested in scheduling the RPi's "get-ON" and "get-OFF" times during the day. How is this possible? How can I set this? Thanks for help.

shawaj, 2020-10-24:
So you would like to turn the pi on and off automatically throughout the day?

For the wake up you can use the wake-up alarm functionality https://github.com/PiSupply/PiJuice/tree/master/Software#wakeup-alarm-menu

And then you can just create a script to shut down the pi as and when you want to.

Hope that helps

mrglobetrotter0088, 2020-10-24:
Ok, thank you. I‘ll give it a try.

shawaj, 2020-10-24:
@mrglobetrotter0088 ok let us know how you get on

shawaj, 2020-12-05:
Closing this as no reply in quite a while. Let us know if you need any other assistance.

closed 2020-12-05 by shawaj