PiSupply/PiJuice #569: What is the easiest way to completely turn off everything?

ovidiurosu90, 2020-10-23:
Hi,

I just got the Pi Hat and it works great.
Something that I didn't understand yet is how to turn off everything.

I disconnect the external power source and turn off the Pi gracefully by running `sudo poweroff` in the terminal.
It seems to fully turn off the Pi as I don't see any LEDs on on the Pi.

The hat is still in standby as the green LED still blinks.
The only solution I found so far is to remove the battery which seems a bit overkill.

Is there a nicer way to completely turn off the hat?

Kind regards,
Ovidiu

shawaj, 2020-10-23:
Have a look here https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#user-functions

You want to use the SYS_FUNC_SYS_OFF_HALT on the long press 1 on SW1

ovidiurosu90, 2020-10-24:
> 
> 
> Have a look here https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#user-functions
> 
> You want to use the SYS_FUNC_SYS_OFF_HALT on the long press 1 on SW1

Thank you for the quick reply. I tried it but it didn't do what I wanted.

I set SYS_FUNC_SYS_OFF_HALT on the long press 1 on SW1. I removed the external power and then I pressed SW1 (for 10s) until D2 LED started blinking. The Pi seems off (though it still has the red LED on) and the hat still has the green LED blinking.

ChristopherRush, 2020-10-24:
You need to use sys func halt pow off for complete power removal to the Pi.

shawaj, 2020-10-24:
Sorry that was my typo there I copied the wrong one... Yes it should be SYS_FUNC_HALT_POW_OFF

ovidiurosu90, 2020-10-24:
Wow, you guys are quick, thank you.

I did the change to SYS_FUNC_HALT_POW_OFF and it's better (Pi doesn't have any ON LEDs).

This is the same behavior I get if I run the following script:
```python
#!/usr/bin/python3

import pijuice
import os
pijuice = pijuice.PiJuice(1, 0x14)

# Remove power to PiJuice MCU IO pins
pijuice.power.SetSystemPowerSwitch(0)

# Remove 5V power to RPi after 20 seconds
pijuice.power.SetPowerOff(20)

# Shut down the RPi
os.system("sudo halt")
```

Almost perfect, though there is still a very dim green LED blinking on the hat (every 3s), which I suppose it's the standby mode. Is there a way to fully turn off the hat? As in, no blinking at all.

Kind regards

shawaj, 2020-10-24:
It should turn off automatically after 60 seconds

shawaj, 2020-10-24:
Sorry just saw you have it set up remove power after 20 seconds in your script. That's fine too.

But the PiJuice will always have power when the battery is in. It will draw about 0.5mA when powered down to maintain the wake on alarm and other functionality

ovidiurosu90, 2020-10-24:
Okay, clear, so there is no way to completely power off everything unless I remove the battery from the hat.
Thank you for clarifying.

tvoverbeek, 2020-10-24:
@ovidiurosu90 You wrote:
```
Almost perfect, though there is still a very dim green LED blinking on the hat (every 3s),
which I suppose it's the standby mode.
Is there a way to fully turn off the hat? As in, no blinking at all.
```
Short answer: No
The PiJuice MCU has to remain active (albeit in low power mode = the dim green blinking) to be able to react on events like 'Charging restart` and '(repeated) alarm wakeups.
If the power completely goes off the firmware loses the RTC and any volatile data.

shawaj, 2020-10-24:
@tvoverbeek looks like we were both replying at the same time :-)

Will close issue now. @ovidiurosu90 if you don't agree then feel free to reply

closed 2020-10-24 by shawaj

FNumidia, 2020-11-16:
> The PiJuice MCU has to remain active (albeit in low power mode = the dim green blinking) to be able to react on events like 'Charging restart` and '(repeated) alarm wakeups.

In the case of a very long period of deep sleep (5 months or more) after which the battery is completely empty we will have the same effect of a firmware reset? In this case could the RaspberryPi wake up it automatically if the power is on again (supposing to have previously set Wake Up on Charge > 10%) or I would need to push the button? Would the battery (BP7X) charge again or it would be seriously damaged?