PiSupply/PiJuice #570: Detecting the SW2 state on PiJuice Zero

Passiday, 2020-10-24:
My project involves Pi Zero and Arduino, powered by PiJuice Zero. There's an external power switch (pushbutton) with LED. Arduino provides some basic funcionality for the periods when the Pi is booting up or shutting down: blinking the power LED to give user feedback about what's going on.

The external power button is attached to PiJuice Zero J5, and since for PiJuize Zero it corresponds to SW2, the SW2 configuration is made identical to that of SW1.

I'd like the Arduino to know when user is pushing the external power button. I know there are PRESS and RELEASE events that can be configured to launch user scripts by PiJuice service, but since the Pi can be in non-booted state, there should be Arduino to handle such attempts.

Using multimeter, I have measured that in power-on state, the J5 pin voltages (using the global GND as ground), when the switch is open, are as follows:
- pin A, the one that is closer to the corner, is -0.07V
- pin B, the other pin that is closer to the middle of side, is 3.23V

When the switch is closed, the pin A raises to 3.23V.

I plan to connect a wire from pin A to Arduino GPIO pin and read it to detect the SW2 state. I wanted to know if that's a safe way to do. I mean, the disconnected voltage for pin A is this weird -0.07V, so perhaps it would be not a good thing to pull it to GND.

tvoverbeek, 2020-10-24:
See the PiJuice-Zero schematic, especially page 2 (https://github.com/PiSupply/PiJuice/blob/master/Documentation/PiJuice_Zero_Schematic.PDF).
You could also connect pin 2 of J5 to an Arduino digital input (then there is a 3K9 resistor between pin A of SW2 and
the Arduino digital input).
The 3.3V when the button is pressed should ne interpreted as HIGH on the Arduino (with 5V logic).

Passiday, 2020-10-24:
Ok, just to be sure — can you, please, confirm the connection diagram:
[image]
I was somewhat puzzled why that pin voltage was not GND, when the switch is open.

ChristopherRush, 2020-10-24:
Could it be an uncalibrated multimeter?

shawaj, 2020-10-24:
@ChristopherRush yeah I think you're probably right. May be a coincidence but it's -0.07 and 3.23 when it should be 0 and 3.3 - so out by the same amount.

@Passiday you can see from the schematic, image below, that it is already pulled to ground...
[image]

Passiday, 2020-10-24:
@ChristopherRush don't know if that's the case. A fresh Duracell AA battery shows 1.6V. Shorting the probes gives 0.0V. If I measure the voltage across the board's GND and 5V holes, I get 5.035V. If I measure the board's GND and J5 pins, I get 0V and 3.3V. So, the problem apparently is that the "global GND" that I used in my original measurement was from one of the Pi Zero pins. For some reason, the Pi Zero GND is 0.07V above the PiJuice's GND. Perhaps there's some current leaking on the board where the Pi is plugged in. But it's obviously not the issue to be discussed here. Thank you all for help.

shawaj, 2020-10-24:
@Passiday so are you happy for us to close the issue?

closed 2020-10-24 by Passiday