PiSupply/PiJuice #571: Running PiJuice Python3 project in virualenv

Passiday, 2020-10-25:
I have PiJuice software running on Pi Zero. Communications with PiJuice from Python3 work fine. But the main Python program of this project is running under virualenv.

When the Python complained with `No module named 'pijuice'`, I manually copied `/usr/lib/python3/dist-packages/pijuice.py` to my project root directory. But then it complained with `No module named 'smbus'`, and `pip install smbus` doesn't resolve nicely, runs into compilation errors, things get hairy.

Is there some recommended way in getting PiJuice Python module to work under virtualenv?

shawaj, 2020-10-25:
@Passiday does virtualenv not have a way to inherit system packages?

Like this https://raspberrypi.stackexchange.com/q/9246/6025

Passiday, 2020-10-26:
Yes, thanks, that did the trick!

closed 2020-10-26 by Passiday

shawaj, 2020-10-27:
Awesome glad to hear it!