PiSupply/PiJuice #572: Prevent power up if USB power not present

blanco-ether, 2020-10-28:
My application uses PiJuice Hat.  The use case is uninterruptible power supply where I only want to prevent abrupt power loss so as to protect OS on MMC.  Power is expected to always be present when in use but can't be enforced if user fails to plug it in.

- Pi 4 or 3B+ with official touchscreen
- 5000mah battery
- Power supply connected to J4 on hat
- SW1 used to power on and power down.
- System event "No power" maps to USER_FUNC1 which is python script that safely powers down (works fine!)
- Pi boots into custom python app which runs in a venv.

I have found that if a user uses SW1 to start the Pi when not plugged into power the battery runs down on them.
Ultimately I would like to to prevent powering the Pi using SW1 unless external power is applied to J4 on hat
but can't find any way to do that from settings.  Is this possible?

Alternately I could notify them within my app that they need to plug in but cannot access pijuice python API because it runs in venv.  Is there a way to install pijuice API in venv.  Do I need to take a different approach and not run my app in a venv?

I could also make settings safely shut down the Pi when battery is low but this is not ideal.

Something I've noticed is that if SW1 starts Pi without power applied then power subsequently applied then removed the 
system event "No power" does not shut down the Pi.

I want to be able to protect abrupt power loss in every case that might occur.

Please advise.

Thank you

tvoverbeek, 2020-10-29:
You wrote:
```
Alternately I could notify them within my app that they need to plug in but cannot access pijuice python API
because it runs in venv. Is there a way to install pijuice API in venv. Do I need to take a different approach
and not run my app in a venv?
```
Start the virtual environment with `python3 -m venv --system-site-packages /path/to/new/virtual/environment`
This should make the system packages available inside the virtual environment.
See https://docs.python.org/3.8/library/venv.html
With this option you could implement the 'please plug-in power' notification in your app using the pijuice API.

Also you wrote:
```
Something I've noticed is that if SW1 starts Pi without power applied then power subsequently applied then removed the
system event "No power" does not shut down the Pi.
```
This is the way the pijuice service is programmed. When the pijuice service is started the internal no-power-count variable is initialized to 300. Every time the no-power is detected the variable is increased by 1. As soon as external power is present the no-power-count is reset to 0. When the count equals 2 the function triggered by no-power is called.
This is to prevent an immediate shut-down when restarting on battery only.

blanco-ether, 2020-10-29:
To be clear, there is no way to prevent powering the Pi using SW1 when external power is absent?

Thank you for indicating a way to access pijuice API from venv.  If I were to notify user that they need to plug in power
but maybe they walked away and didn't see message I could safely power down after some time.  That is covered.

However they might have noticed the message and followed the recommendation and plugged into power
supply.  At that point, if they were to accidentally remove power (tripped over cord, whatever)  and walk away the auto power shutdown would not work as they applied power AFTER powering up.  Is there any way in API to reset this event
in this case?

Alternative might be to constantly poll power present through pijuice API in my app and always warn that if they don't plug in it will power down automatically after some time.

Our application is to run test fixture for circuit boards.  It causes no harm if power safely shuts down the Pi when it is 
unattended without external power source.

I am trying to cover all possibilities.  It may be helpful to establish recommended practice as it is possible others rely on pijuice for uninterruptible power similar to the way I am in their own application.

Thank you for your help.

tvoverbeek, 2020-10-29:
You could reprogram SW1 in case of start without external power. However this also prevents the start when external power is back (unless you use the wakeup-on-charge, bu tthat has its own complications)
I think  simplest approach is to check for external power at the start of your app, give the message/notification and shut down again. That way external power has to be present to be able to pass this initial check.

blanco-ether, 2020-10-29:
After my last post I have thought it through and agree that best approach is to simply monitor external power present in
my app at all times and notify user when not present that auto shutdown is imminent after say 1 minute if user fails to take action.  We always require power and so we can reasonably count on battery charge.

Thank you for your help.

I am grateful that you came up pijuice hat, it is ideal for our and many others needs.  So much so that it seems to be in high demand and sometimes hard to get in US not to mention demand seems to make the price go way up.

Nice work guys!!

closed 2020-10-29 by blanco-ether