PiSupply/PiJuice #573: Set the time from no power detected to system event fires

blanco-ether, 2020-10-30:
Greetings,
Can I set the amount of time it takes for system event "No Power" to fire after power loss.

When RPi boots it automatically starts my app.  My app monitors power loss through
PiJuice python API and if lost notifies user that shutdown will occur in 30s unless power is restored.

While developing the app on the RPi there are periods of time that the app is not running and
I would still like to automatically safely shut down if power were to be lost in this case.
In order to make it all work correctly I need to make "No Power" event fire some time
after the count down in my app.  Is there a way to do this?

Thanks

tvoverbeek, 2020-11-04:
If I understand you correctly your app does the main monitoring of the power and shuts down the Pi after 30 seconds.

Suggest to also enable the system 'no power' event and link it to an other user function which delays the shutdown for say 35 seconds. Of course, just before shutting down this function should check it the power is still absent and if not just exit without any action. This way it will not interfere with the power monitoring in your app.

blanco-ether, 2020-11-05:
Yes, I had thought about such a solution.  I looked through documentation for 
pijuice_cli and GUI as well as Python API and did not find anything that looked like it applied
to that timing so thought I would ask anyway.  Just for curiosity, what is the time interval
for system event to fire when power lost.  Empirically I measure about 3-4s.  Is it
timed exactly in pijuice firmware or does it depend on other factors in circuit?

Thank you

tvoverbeek, 2020-11-07:
The actual power monitoring is not done by the pijuice firmware, but the pijuice service running on the Pi.
When the service is running and the no power event is enabled the service (pijuice_sys.py) polls the power inputs approximately every 5 seconds. It needs 2 consecutive no power polls to start the function connected to the event.
See the source https://github.com/PiSupply/PiJuice/blob/master/Software/Source/src/pijuice_sys.py lines 321-343.
The service does more or less the same as your app.