PiSupply/PiJuice #574: GPL Licence

DazFahy, 2020-11-02:
Hi, would it be possible to update the documentation to include the command addresses which are used to communicate with the PiJuice and also any other missing documentation.

Basically I am wanting to implement a C# version of the code in a non GPL compatible code, it is an MIT licence. However because I have had to read the code to obtain some of the functionality, the other project are saying that because they are MIT and the code I have read is GPL then they cannot merge my code for legal reasons.

shawaj, 2020-11-03:
@DazFahy what is the other project you are looking to put it in?

Perhaps you can suggest something we can add to the readme, which would serve the purpose you are mentioning? As you will know better than us exactly what you need.

DazFahy, 2020-11-03:
@shawaj it is the Microsoft [dotnet/iot](https://github.com/dotnet/iot) project which is for IoT device bindings for .Net Core on Raspberry Pi's and other boards.

Because they are on the MIT licence they have said that in order for the code I have written to be added to the repo, I cannot have used the source code from your repo because you are on a GPL licence. They have said that the only way is if the documentation provides the detail which can be used to write the code.

shawaj, 2020-11-03:
Just for reference this relates to https://github.com/dotnet/iot/pull/1213

@DazFahy that is fine. However, what I am trying to understand is exactly what info you need to be in the documentation. Could you elaborate on what exactly you need documenting to get this resolved?

And are you perhaps able to suggest (perhaps via a PR or just a comment in this issue) what this documentation should / could look like?

tvoverbeek, 2020-11-03:
@shawaj He needs all the details of the low level I2C commands and responses. Now the only way to get those details is looking at the pijuice.py module source code and implement it in c#. For legal reasons he does not seem allowed to do this since our code is GPL We would have to document the low level communication in a programing language neutral fashion (like e.g. the I2C communication in data sheets).

shawaj, 2020-11-05:
@tvoverbeek do you think this is something feasible to do?

DazFahy, 2020-11-05:
@shawj and @tvoverbeek I am happy to help produce any documentation

shawaj, 2020-11-05:
@DazFahy if you're happy to that would be great :-)

AmedeeBulle, 2020-11-22:
On a side note, `pijuice.py` is licensed GPL V2, while the project LICENSE file states GPL V3...
Can you clarify?

Note that it would help to have the `pijuice.py` module itself licensed LGPL to avoid issues when it is imported in a script with other python modules which might have an incompatible license...
(E.g.  you can't use GPL V2 and Apache V2 modules in the same script...)

shawaj, 2020-11-22:
@AmedeeBulle I guess you mean setup.py not pijuice.py?

AmedeeBulle, 2020-11-22:
Indirectly...

The [license stanza](https://github.com/PiSupply/PiJuice/blob/master/Software/Source/setup.py#L62) of `setup.py` states the license of the packaged python module which is `pijuice.py` for the `pijuice-base` package.

shawaj, 2020-11-23:
@tvoverbeek was that a mistake re @AmedeeBulle comment on the license? Are you happy for us to change to GPL v3 to be the same as the rest of the repo?

tvoverbeek, 2020-11-24:
@shawaj whoever wrote the original setup.py (Francesco or Milan?) put in GPL v2. Might indeed be a good idea to license pijuice.py as LGPL so ir can be used in programs with other licenses as long as they do not modify pijuice.py.

shawaj, 2020-11-24:
@tvoverbeek I'm happy to change the whole license to LGPL or MIT or something more permissive. No particular reason for choosing GPL v3 really.

shawaj, 2020-11-26:
Update here based on email from @richlander - https://github.com/dotnet/iot/pull/1213#issuecomment-734434507

richlander, 2020-12-01:
Seems like this issue can be closed now.

shawaj, 2020-12-01:
Thanks @richlander 

@tvoverbeek any further thoughts re changing to LGPL or MIT or something like that? Do you think it would be worthwhile?

shawaj, 2020-12-05:
Also @DazFahy not sure if you would still be happy to write up any stuff about what you did to convert this? 

Could be really helpful for people in the future 🙂

DazFahy, 2020-12-14:
Hi @shawaj, yes definitely. I have some time between now and the new year, I will try and come up with a formatting example that we can agree on for the information. Will probably need some help though on some of the descriptions on what the code is doing though.

DazFahy, 2020-12-15:
Hi @shawaj, below is an example of what I'm thinking it might look like. I have based this on the [BMP280 datasheet](https://cdn-shop.adafruit.com/datasheets/BST-BMP280-DS001-11.pdf). I've chosen the Firmware version due to it being one of the simpler pieces of code to document. 

Let me know what you think, any other suggestions for the format or contents are much appreciated.

# PiJuice
## Register 0xFD "firmware"
The "firmware" register contains the firmware version of the pijuice.

### Register 0xFD Byte 0 "version"
|  Register 0xFD "version" |    Name    |             Description             |
|:--------------:|:----------:|:-----------------------------------:|
| Bit 7, 6, 5, 4 | major[3:0] | Contains the Major Firmware Version |
| Bit 3, 2, 1, 0 | minor[3:0] | Contains the Minor Firmware Version |

### Register 0xFD Byte 1 "variant"
|  Register 0xFD "firmware variant" |    Name    |             Description             |
|:--------------:|:----------:|:-----------------------------------:|
| Bit 7, 6, 5, 4, 3, 2, 1, 0 | variant[7:0] | Contains the Firmware Version Variant |

shawaj, 2020-12-15:
@DazFahy sure, happy to help and I'm sure @tvoverbeek and @ryanteck and @mmilann will be as well!

With regard to the formatting... It looks absolutely fantastic to me! Nice one.

mmilann, 2020-12-15:
Variant is about hardware compatibility and currently only one variant is
there. For example it can be checked before updating to suitable firmware.

shawaj, 2020-12-16:
@mmilann just so you know - @DazFahy built this https://github.com/dotnet/iot/pull/1213

And has very kindly offered to help us document the firmware functionality and info so other people can build software support without relying on our GPL licensed code.

If you can provide any further assistance on this @mmilann it would be much appreciated.

mmilann, 2020-12-16:
@DazFahy suggestion looks nice, I would just change naming. Instead of "Register" more suitable is to say "Command code" because it is not actual contiguous memory read but more like remote procedure call.