PiSupply/PiJuice #577: PiJuice Zero UPS hat, battery discharging, general wired behavior

kaulex99, 2020-11-07:
Hardware Setup:

- Raspberry Pi Zero W
- Pijuice Zero UPS hat
- 6W Pijuice Solar Panel
- 5000mAh LIPO with NTC

What I did:

1. Put the hat on the pi
2. booted & installed pi juice software with gui
3. set battery to 5000mAh
Battery: 
[image]
Current status:
[image]
4. Now I plugged in the battery and the solar panel
Current status: 
[image]
Status after a view **seconds**
[image]

As you can see the battery is discharging. I really don't know why. Solar Panel is outside on the window. (also getting a little bit hot in the sun) 
Nevertheless, the temperature of the battery is not as hot as the stats say. In reallife battery has a temp of 25-30°, stats say 91°.

Also the % of the battery is jumping fast. After a reboot the temp. is correct for a short time but then it jumps to 91°, really don't know why.

If you need further information, don't hesitate to ask.

tvoverbeek, 2020-11-07:
A few things:
- Both GPIO power input and USB Micro power input show PRESENT. This is strange.
  In your setup I would expect only the USB power input to be present (assuming the solar array output is connected to the 
  usb micro of the PiJuice Zero 
- The varying temperatures was a known issue in the previous version of the firmware. Which firmware version are you
  running? See the firmware tab in pijuice_cli. Current version is 1.4.
- Also from the status screens it is clear the battery is not charging (battery state NORMAL, not CHARGING FROM ...)

Could you provide a picture of your setup, especially the connection between the battery and the PiJuice Zero?
Which version of Raspbian/Raspbery Pi OS are you running.
What version of pijuice-base package? (Use `apt policy pijuice-base` to find out).

kaulex99, 2020-11-07:
@tvoverbeek 

- I forgot to say: I had connected it via a normal power supply too otherwise the pi immediately shutdown. Therefore, the GPIO is present.

- Firmware is up to date with 1.4

- yeah

`pi@raspberrypi:~ $ sudo apt policy pijuice-base
pijuice-base:
  Installiert:           1.6
  Installationskandidat: 1.6
  Versionstabelle:
 *** 1.6 500
        500 http://archive.raspberrypi.org/debian buster/main armhf Packages
        100 /var/lib/dpkg/status
`

2020-08-20-raspios-buster-armhf

For sure: 
[image]

shawaj, 2020-11-07:
@kaulex99 which battery are you using? Is it one you bought separately or one of ours?

Is the battery connector pushed on as far as it can go? It looks like there's a bit of a gap?

Could you share a picture of the battery also?

kaulex99, 2020-11-07:
No i bought that one extra but is was also named PiJuice 5000mAh battery 

[image]

tvoverbeek, 2020-11-07:
This all looks fine.
In pijuice_cli can you check on the General tab the status of 'Charging enabled' and 'GPIO input enable'.
I suspect that charging is not enabled.
From the battery parameters you can see the charging current in the constant current charging phase is 2.5A
Your solar panel can only deliver max 1A at 5V at its usb output. So charging will take longer.
If you use your normal power adapter (connected to the Pi zero as in your picture) and disconnect the solar array see if you can charge the battery that way. That would at least confirm charging works.

kaulex99, 2020-11-07:
I will try that, but this don't explain why the temperature sensor is measuring such a high value.

kaulex99, 2020-11-08:
Not charging with a 1A power adapter over GPIO
[image]
I have also added another power adapter on the Pi hat, not charging either.
[image]

kaulex99, 2020-11-08:
Interesting thing happened just now. I disconnected the battery and connected it again. Suddenly it is charging, 1...2...%, also temp is ok with 25°.

After 3s it jumps to 91° not charging anymore. 

I did that once again battery now 99% charged wich is obviously not possible. 
[image]

After a view more seconds 
[image]

shawaj, 2020-11-08:
@kaulex99 a 1A power adapter is probably not sufficient to power the Pi and charge the battery. You should ideally be using a 2.5A or 3A power supply.

What you've described sounds like possibly a loose connection somewhere. Does moving the battery around cause any similar changes? I wonder if there is a loose cable.

Do you have another battery you can test with also?

kaulex99, 2020-11-08:
@shawaj Sadly, I don't have another battery. I could order one in case we don't have another solution.

Moving the battery around does not help too.

I thought the Pi Zero is not consuming that much energy, anyway, I tried a 3A one and it didn't work too 😢 I am running out of ideas maybe the battery is faulty as you mentioned.

tvoverbeek, 2020-11-08:
If the firmware believes the high temperature it will stop charging the battery.
Go to the Battery Profile tab in pijuice_cli and scroll all the way down.
Change the "Temperature sense" to NOT_USED.
The firmware then will see a fixed temperature of 25 deg.
Then with the larger power supply uou should see charging.
You definitely seem to have a problem with the NTC battery temperature measurement.

shawaj, 2020-11-08:
@kaulex99 where did you buy the battery from? I think it would be best to organise a replacement but depending on where you got it from changes the best way to do that

kaulex99, 2020-11-08:
Bought it from [Distrelec](https://www.distrelec.com). (Austria)

shawaj, 2020-11-08:
Can you send an email to sales@pi-supply.com with your address and a link to this issue and you purchase receipt. I'll get a new one shipped out to you.

kaulex99, 2020-11-08:
Thank you. I have already sent the email.

shawaj, 2020-11-08:
Just replied with tracking @kaulex99 you should receive it Tuesday or Wednesday I think.

Hopefully it might fix the issue. Let us know either way :-)

kaulex99, 2020-11-11:
Received new battery today. Sadly, it is not working either :(((( now tmp is about 250°
[image]

kaulex99, 2020-11-11:
and it is discharging too.

kaulex99, 2020-11-11:
I have recorded it too
Starting with battery connected -> then disconnect -> then connect again
[image]

shawaj, 2020-11-11:
@tvoverbeek what do you think? Maybe the PiJuice Zero itself is the issue?

tvoverbeek, 2020-11-11:
I would still like to know what happens if @kaulex99 puts the "Temperature sense" to NOT_USED.
See my comment above (https://github.com/PiSupply/PiJuice/issues/577#issuecomment-723582039).
Something is wrong with the temperature measurement.

kaulex99, 2020-11-11:
Oh sorry, I forgot that.

kaulex99, 2020-11-11:
@tvoverbeek charging as it should (disabled ntc), I will try it tomorrow with the solar panel too. Nevertheless, I use a 3A power supply now and it still weak sometimes, don't know why. Any ideas? 

Furthermore, I checked the battery connector on the board with my multi. All seems connected fine. 
[image]

tvoverbeek, 2020-11-11:
Note that the charge current for the 5000mAh is 2A. Then there is the overhead of the power conversions on the PiJuice Zero and with a RPi consumption of 1A it could be marginal from the batteries point of view.
A bit suspicious is the GPIO power input voltage. It is below 5V. Now it is charging it shows only 4.84V. When not charging it shows 4.96V. Smells like the 5V leads of your power supply have a too high resistance for the current asked.
Here is my screen (for the default BP7X 1820mAh):
[image]

Anyway there is something wrong with measuring the temperature with the internal NTC of the battery.
Since you have the same problem with 2 different batteries it is probably a bad contact or a pcb trace on the PiJuice Zero.
@shawaj Maybe time to offer @kaulex99 a replacement PiJuice Zero???

shawaj, 2020-11-11:
@kaulex99 can you let me know the support ticket number from previous email you sent?

kaulex99, 2020-11-11:
@shawaj Sure: 119939

kaulex99, 2020-11-11:
@tvoverbeek looks like the voltage is ok currently.
[image]

tvoverbeek, 2020-11-11:
This is with the solar panel? Looks ok. 0.158A is the current (not an actual measurement, but an estimate) going to the Pi.
Note the voltage 5.09V which is the output from the 5V converter on the PiJuice Zero.

kaulex99, 2020-11-11:
No, I will try the solar panel tomorrow. Currently it is night in Austria so I don't have a good lighting situation :D

tvoverbeek, 2020-11-11:
@kaulex99 I can confirm it is dark here. I am living in Fieberbrunn, Tirol.

kaulex99, 2020-11-12:
@shawaj Shall I get in touch with you via email?

shawaj, 2020-11-12:
@kaulex99 have emailed you. Replacement PiJuice Zero shipping out tomorrow.

kaulex99, 2020-11-13:
@tvoverbeek out of question:  Do you think it is possible to run a pi zero with 5000mAh battery and a 22W solar panel 24/7 in austria or do I need a larger panel? 

@shawaj Thank you!

tvoverbeek, 2020-11-13:
Depends on what your pi zero (not zero w??) is doing 24/7. Makes a huge difference if it is on only a few minutes per hour or continuously. A full 5000 mAh battery will not last 24 hours when the pi zero is on continuously.
When it is off for a considerable time then the solar panel has a chance to charge the battery.
Have you seen the battery discharge calculator https://learn.pi-supply.com/battery-levels/?
Anyway you definitely need a larger solar panel to get a charging current of > 2A for the periods when the pi zero is off.

kaulex99, 2020-11-13:
@tvoverbeek overbeek zero w yeah, forgot the W sorry.
The site says 23.13 hours. With a solar panel which produces enough power, this should work. 

What do you mean with off? It should never be off. It should run 24/7.

tvoverbeek, 2020-11-14:
You wrote;
```
What do you mean with off? It should never be off. It should run 24/7.
```
A scenario like this: a few measurements every 5 minutes. Most of the time the Pi Zero will be idle then. Using repeated wakeup you could shut down the Pi after the measurements are done (and stored or sent somewhere). During the off time the solar panel can charge the battery much more eggectively.

Your solar panel calculations do not take into account the conversion losses to get to the 5V output. As usual solar array capacity is the raw output from the panel, not the regulated output. Moreover there are also conversions on the PiJuice so you might lose 20 to 30%.

kaulex99, 2020-11-17:
@tvoverbeek @shawaj  new pi hat 😢 still 248°
[image]

shawaj, 2020-11-17:
Hmm very strange. Do you have another Pi, SD card and power supply you could try?

kaulex99, 2020-11-18:
Yes I will replace everything and check it again. Really strange

shawaj, 2020-11-18:
@kaulex99 thanks very much keep us posted

kaulex99, 2020-11-20:
Update: New SD Card with fresh raspian os and newest version of pijuice-base 
[image]

kaulex99, 2020-11-25:
@shawaj finally, everthing is working - I checked the pi hat with another power supply and another pi (PI 2B+) and it is working like a charme. Today I'll get a new raspberry pi zero w too. Then I can check if the problem is generally with the pi zero.
[image]

kaulex99, 2020-11-25:
@shawaj had to delete my previous comment. It is not working with another pi and another power supply of 3A. (PI 2B+)
still 240°.

kaulex99, 2021-06-03:
To close this issue: Firmware Version 1.5 solved the issue with the temperature.

Thank you for you help.

closed 2021-06-03 by kaulex99