PiSupply/PiJuice #579: STM32F030CCT6 alternatives?!?

shawaj, 2020-11-12:
There is a current supply issue with STM32F030CCT6 affecting most ST suppliers. The only chips available are changing hands for $4+ a chip as opposed to $0.75 ish at 2k pcs.

Think it is worth exploring some alternatives.

STM32F091CCT6  looks pin compatible, and it also has all features and I2C bootloader for firmware update over i2c. However it will need firmware rework/reconfiguration (price is over two times STM32F030CCT6 however for emergency use could be a good alternative)

Another part that looks interesting is 64 pin version STM32F030RCT6 is firmware compatible but needs PCB update, and maybe also hard to source.

I think best option is to go pin compatible to avoid having to have two different PCB versions...more versatile. Although we could do both - and then we have 3 options STM32F030CCT6, STM32F091CCT6 and STM32F030RCT6.

@tvoverbeek @ryanteck do you have any thoughts / concerns about this?

ryanteck, 2020-11-12:
I was possibly going to suggest the STM32F030C6 which was the one I used on the RTk.GPIO and from the same family so should have been pin compatible. But it seems that every variation of the family has shortages currently at farnell indeed ranging from delivery from Febuary 21 to even next November!

I did think about the slightly updated ST chip as to weather that's pin compatible or could easily be adjusted I wasn't sure. But was the one I've got designed in for the I2C GPS LoRa addon. However farnell are indicating after their existing stock of that one it could face the same issue.

The 3 options suggested are all currently out of stock at Farnell for similar dates, so it makes me wonder is it an ST issue and would we possiby have to look at a different brand?

shawaj, 2020-11-13:
@ryanteck that could potentially be a good idea - a different chip - my only concern with that would be how much complexity it adds to the firmware update process in the software. But I guess we can deal with this by auto-detecting which chip is on the specific board and only offer the relevant firmware.

I wonder if there are any other chips in a LQFP-48 package that can be configured as pin compatible from another manufacturer such as NXP / Nordic / Microchip or someone else. Worth investigating I guess. As you say this might make things more resilient to have from different manufacturers.

ryanteck, 2020-11-13:
Possibly altough unfortunately I don't think there will be a chip that will be pin compatible exactly.

ryanteck, 2020-11-13:
https://www.theburnin.com/industry/stmicro-temporarily-halt-french-semiconductor-production-covid-19-coronavirus-2020-3/

shawaj, 2020-11-13:
That explains it then. Good find @ryanteck

shawaj, 2020-11-14:
I wonder whether something like a gigadevices GD32 might be interesting and better value overall.

https://www.gigadevice.com/products/microcontrollers/gd32/

They even have a RISC-V version 😁

shawaj, 2020-11-14:
GD32F101 series and GD32VF103 series look interesting

ryanteck, 2020-11-14:
I have to say I've never heard of that brand, is it a Chinese one?

tvoverbeek, 2020-11-14:
See their website. They are in Silicon Valley USA. What is not clear where the foundries are who produce the chips. Looks like China (Wuhan). See http://www.xmcwh.com/index.php?s=/cms/101.html

shawaj, 2020-11-14:
@ryanteck @tvoverbeek there is also http://www.wch.cn/products/category/5.html

And yeah giga devices are headquartered in Beijing I think. Their website says silicon valley but I think that may just be for show perhaps.

ryanteck, 2020-11-15:
I've used a couple of wch chips before with success. 

Only downside of both is not as easily available from Farnell, RS, Mouser, Digikey, Arrow & Avnet etc. But it's not like the ST currently are either.

Unfortunately it doesn't look like the firmware was written using mbed which is fairly good for multiple chip support. So it'd require a full re-write with the new chip.

shawaj, 2021-02-25:
The ambiq Apollo range looks interesting @mmilann