PiSupply/PiJuice #580: PiJuice Hardware Updates - USB C etc

shawaj, 2020-11-12:
Starting this issue to track hardware updates of PiJuice HAT / Zero surrounding the Pi 4 upgrades - USB C etc.

**These are already included updates of PiJuice and Zero boards from last year, but not finished...waiting for confirmation to be released (see attached images):**

- LED off-board header added
- Removed unreliable fuel-gauge IC. Will use firmware to measure battery charge and temperature (already exists in V1.4 firmware)
- USB -Type C connector
- LED color mixing improvement with independent driving
- Added opamp for more precise current measurement of 5V GPIO current to Rpi
- Silkscreen updates
- Added run pin location for Pi4

**These are other proposed updates:**

- Potential update to add battery connector on HAT like on zero (probably under PCB so we do not need to redesign the battery surround parts)
- uprate for 3A input / output
- rework for STM32F030CCT6 alternatives - see #579 also 
- support for multiple cells / other battery technologies
- support for wider input voltage (up to ~16v or 26v for car / truck usage

**Miscellaneous:**

- change battery capacity storage in firmware to uint32
- any other firmware / RTC / other updates?

**Images...**

[image]
[image]

@tvoverbeek @ryanteck any other ideas / thoughts on potential hardware updates / upgrades?

shawaj, 2020-11-12:
There is also this firmware issue from @tvoverbeek email of 27/07/20...

> With reference to the email conversation with Jean-Christophe Poisson about his
> problem with the 1.4 Firmware (repeated wakeup not working when solar array input present
> with 1.4 firmware, while 1.3 works)

ryanteck, 2020-11-12:
All the main suggestions look good to me, USB C would be nice indeed.

I think either adding a connector or replacing the screw terminal connector with the same one as the Zero would be a great idea as it makes it that bit more user friendly for the bigger batteries we sell.

I think possibly 3A output would be quite good, the 2.5A input is likely fine as for the surges the battery can compensate. But a slightly higher output would be beneficial.

For the input voltage increase we were looking at (and I was mid way through designing) a Pi Juice wide input pHAT which took up to I think 19V in and regulated it down to 5V 2A for the GPIO header and could be used alongside or independantly of Pi Juice. If we integrated it on the board this additional add-on wouldn't be needed so it could be good to decide if we split it off as it's own thing or merge it in if we're re-designing.  (I went with a 19V target to support the 12-15V from a car 12V socket, but also to then accept the 19V DC Jack output of our 40W solar panel).

We have had some request for multiple cells, however I'm not sure how easy it would be to add this in.

shawaj, 2020-11-13:
@ryanteck do you think USB-C on the PiJuice Zero as well? Guess it could be good to have that and then you have option of either USB micro or USB-C perhaps?

Was thinking of the other battery connector to be as well as the screw terminal - firstly because moving that compromises the battery surround setup. But also because it is easier for a non-PiJuice branded battery to use the screw terminals.

Are there any other features in the software / firmware / hardware that you would like to see in an update?

ryanteck, 2020-11-13:
I think USB-C on the Zero too, but would be interesting to see what others think.

Yeah both battery terminals would be good, we'd need to maybe try and make it so it's clear not to have a battery in the main holder + the socket underneath in that case.

shawaj, 2020-11-13:
@ryanteck yeah i agree about usb c on zero

and yeah i think we just have to clarify not to plug two batteries in at once. Inevitably someone will attempt to though. Maybe there is a way we can physically / electrically stop this somehow but not entirely sure how - maybe one for @mmilann

mmilann, 2020-11-13:
It would not be necessary, it is just initial stress if one is full, other
empty connected in parallel, additionally there will be two NTC in parallel
so reported temperature will be too high and charging disabled as kind of
protection. Further protection with automatic switch is possible but will
degrade performance with additional power/voltage loss on switching
transistor, so not worthy for rare cases.

shawaj, 2020-11-13:
@mmilann ok great - not an issue then in that case. We can just put a note about it on here.

shawaj, 2020-11-14:
I wonder if #425 is worth looking into further also

shawaj, 2020-11-14:
Another issue worth looking into is #357 (even though it is currently closed). I'll try and go through all the open issues this weekend and see if I can find anything else as well that might be worth looking into

devicenull, 2020-11-15:
12v input would be great for me - I'm using pijuice to provide graceful shutdown when my car turns off.  Eliminating the extra power supply there would certainly make things easier.

I had mentioned it in #567 - but the ability to run off of one or more 18650's would also be great.  They're a lot easier to find then arbitrary cell phone batteries or somewhat non-standard battery packs.

shawaj, 2020-11-15:
@devicenull thanks for the input.

We may not put the 12v functionality on the main board, but @ryanteck has been working on a design for an add on that sits on top for this so will be achieved one way or another.

Exploring the 18650 one too 🤩

shawaj, 2020-11-15:
@mmilann we should also add any updates to the PiJuice HAT to the PiJuice Data as well.

We will probably aim to launch it soon but may as well do with the most up to date hardware.

shawaj, 2020-11-15:
Just had a thought that we should also look to address #541 and #423

tilghman, 2020-11-16:
@shawaj You mentioned adding a run pin location for the Pi 4.  Would it be possible to include a socket on the PiJuice for the run pin, so that those of us who are less than comfortable with soldering could potentially just insert a run pin into that socket on the PiJuice?

shawaj, 2020-11-16:
@tilghman I am unsure what you mean here? You can already do this without soldering in the latest firmware using GPIO3 https://github.com/PiSupply/PiJuice/blob/master/Software/README.md#wakeup-alarm-menu

tilghman, 2020-11-16:
@shawaj Is that also true for the Pi 4?  The documentation is written from the perspective from before its release, so it's not exactly clear.

shawaj, 2020-11-16:
@tilghman yes you don't need to use the pogo pin on any Pi anymore to reset whilst powering from Pi micro USB. You can go that with gpio3.

The pogo pin is now optional and we don't include it in the package.

mmilann, 2020-11-16:
RUN pin can be useful for watchdog reset.

mmilann, 2020-11-16:
I would not change USB micro to USB-C on Zero because of higher cost of connector >£0.5.

shawaj, 2020-11-16:
@mmilann ahh ok - maybe not then. Guess can use USB-C adapter if necessary. It doesn't need the extra power anyway.

shawaj, 2020-11-16:
@tilghman https://github.com/PiSupply/PiJuice/issues/580#issuecomment-728168837

ryanteck, 2020-11-16:
> @mmilann ahh ok - maybe not then. Guess can use USB-C adapter if necessary. It doesn't need the extra power anyway.

The main benefit would be the convenience of USB C now rather than power. It would be nice to have if possible because then it's only one cable needed no matter if you're using the Zero or Full Hat. It feels like a shame the Zero wouldn't have it if the full size one has.

shawaj, 2020-11-16:
@ryanteck yeah I guess so. That's a fair point. And I guess removing the LC709203F saves about 0.75 USD as well so evens out.

mmilann, 2020-11-16:
@ryanteck yes that is true, can facilitate production together with hat version. I expect Aaron to dig some $0.01 ch sourced usb c.

shawaj, 2020-11-16:
@mmilann haha thanks for the challenge! Do you have part number of USB-C connector used on this? I will check for alternatives with Sunsoar.

mmilann, 2020-11-16:
@shawaj USB4105-GF-A it is one of cheaper optimised for chargers.

shawaj, 2020-11-17:
@mmilann there is these two attached. 

[U CTY-F-16-30N-WRS-P-S1-F-X-000.pdf](https://github.com/PiSupply/PiJuice/files/5550972/U.CTY-F-16-30N-WRS-P-S1-F-X-000.pdf)

[（已压缩）TYPE-C 16PIN 母座(040）.pdf](https://github.com/PiSupply/PiJuice/files/5550973/TYPE-C.16PIN.040.pdf)

usd0.05/pc for more than 10000pcs, MOQ 1000PCS/reel... For second one. Waiting for the price on first one.

shawaj, 2020-11-17:
@mmilann for first one

Quantity: 2000
Unit Price: USD0.055/pcs
 
Quantity: 10000
Unit Price: USD0.05/pcs

Are they suitable?

mmilann, 2020-11-17:
@shawaj can't believe that price. Order em all before exhausted, they are same dimensions.

shawaj, 2020-11-17:
@mmilann me too. I thought they made a mistake with it and should be 0.5 but they said it is correct

shawaj, 2020-12-05:
Copying this over from #584 so we have it all in one issue...

```
So my question is: Would it be possible for future PCB's to have test-pads for the SDA and SCL line of the I2C-2 bus?
Also a test-pad for the SW1, SW2, SW3, IO1, IO2 would be really nice for me.
If there is any space left also for all three channels of both the LED's.
```

So basically add a bunch of test pads for various things :-)

shawaj, 2020-12-05:
We should also look into #140 as well @mmilann as it seems like a good idea...

> Currently PiJuice address in GUI and System task sources is hard-coded to default 0x14. If HAT address is changed from HAT configuration window PiJuice becomes non-responsive to software and only way to recover is to reset configuration by button press. In future update there should be kind of saving currently configured address (when it is changed from HAT configuration window) into config file, for example there can be two json entries:
address1:0x14
address2:0x68
Additionally there can be resolve function in case those entries in json are for some reason corrupted. Some kind of charge level / status polling in range of I2C addresses similar to i2cdetect, success will be response with correct checksum, triggered with button press next to address change entry in HAT configuration window.

kfandersen01, 2020-12-08:
Can you update the schematics pdf files as well? The connector P1 has mirrored pinout:
[image]

shawaj, 2020-12-19:
@mmilann ^^

shawaj, 2020-12-19:
See #590 also

mmilann, 2020-12-28:
There is smt variant of pijuice zero battery connector, can be on bottom without need to change layout on top. What position is better 1 or 2?

[image]

tvoverbeek, 2020-12-28:
Position 1 is bad. The battery connector is then on the side of the USB connectors on the RPi. Will be nearly impossible to get a battery cable connected there.
Position 2 is OK. All power connections (RPi, PiJuice USB and PJZero battery on the same side.

shawaj, 2020-12-28:
Yeah I agree with @tvoverbeek position 2 is better otherwise you have to remove the hat entirely to plug in a battery.

mmilann, 2020-12-28:
Yes, although situation is better on Rpi 4 because there is around 5mm for cable out between board and USB connectors.

shawaj, 2020-12-28:
@mmilann true but I will think it is not so good because we will not do two versions - we will have this as only version for all Pi versions and also it is still more difficult to plug than facing other day. Without removing the hat from Pi

STrRedWolf, 2021-03-15:
Just subscribing to this.  There's a bit of a surge with Raspberry Pi based "cyberdecks" and having a PiJuice with a 3A capability will let folks build them with just a display, a Pi 4 8gb or Pi 400, a display, and some printing.

shawaj, 2021-03-15:
@STrRedWolf we are making good progress but don't have a timeline on finished units coming to market yet. But we will be sending off for prototypes in the next couple of weeks.

mmilann, 2021-03-22:
Alternative I2C GPIOs. When DPI interface is enabled GPIO2/3 are taken. There can be jumpers to switch to other free GPIOs and software I2C. @ryanteck @tvoverbeek @shawaj  What would be best choice for alternative pins?

tvoverbeek, 2021-03-22:
@mmilann See https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/
RGB888 takes all pins.
All the other DPI modes leave GPIO 26 and 27  (pins 27 and 13) free.
Suggest to use those for jumpers and SW I2C

mmilann, 2021-03-22:
Yes, according to table 26/27 pair covers all modes and there could be even additional SPI pins free in 565 configuration.

shawaj, 2021-03-30:
Sounds good to me @mmilann  @tvoverbeek re the pins

shawaj, 2021-03-30:
Incoming changes...

[image]
[image]
[image]
[PiJuice HAT Change log.txt](https://github.com/PiSupply/PiJuice/files/6227747/PiJuice.HAT.Change.log.txt)
[PiJuice Zero Change log.txt](https://github.com/PiSupply/PiJuice/files/6227748/PiJuice.Zero.Change.log.txt)
[PiJuice Data Change log.txt](https://github.com/PiSupply/PiJuice/files/6227749/PiJuice.Data.Change.log.txt)

mmilann, 2021-03-30:
Lacking space for additional TH header for external option D2/SW2, instead there are test points TP32,TP33,TP34,TP35.

alpapak, 2021-04-03:
I would like to see a hat(like PiJuice, PiBattery) on top or bottom of RPi with standoff screws and a wire to connect to PiJuice Hat. So, it will be a nice add on, stackable, instead of the external battery or the small internal. Also, It could be a two 18650 holder and extended to 2 or more hats(one on top of the other). I think this one, is easy and most popular at the moment. Nowdays, more mAh are needed for projects/peripherals.

devicenull, 2021-06-01:
If it's not too late - some sort of mechanical attachment for the battery would be nice.  The friction fit it currently has works ok for somewhat stationary usage, but not bouncing around in the back of a car.  I've tried hot glue, but it's not super effective (especially when things start to heat up)

Any ETA on availability?  I know supply chains are completely screwed up now, but I'd love to see a new hardware revision here (I think this is one of my last USB micro devices)

leighghunt, 2021-12-09:
Hi @shawaj I'm coming into this quite late compared to everyone - been putting together some kit for some remote long-term timelapse cameras with a Pi High Quality camera - and the one thing I didn't anticipate being so as difficult to source as it's been is LiPo batteries!

Granted, it's probably not helped by me being in NZ and Covid and all - I've managed to find one place that's willing to ship larger LiPos - but is something like the 18650 supported by the PiJuice Zero? I could get these locally tomorrow vs waiting 4-8 weeks for something to be shipped from China.

Thanks for the great project.

cc @devicenull @alpapak

leighghunt, 2022-01-03:
> ... but is something like the 18650 or 26650 format supported by the PiJuice Zero? 

And to answer my own question in a non-official - no-blame-if-it-fries-your-board kind of way, it appears that the 18650 works perfectly fine with the Pi Juice. I'm using a PowerTech 2600mAh 18650 battery with protection circuit - model SB-2299, and it is happily powering the Pi, and being charged by solar.

Power draw is around 175mA whilst being continuously powered, averaging about 90mA if powering on for 1-2 min in every 10 and taking a picture, and 20mA if powering up once an hour and just logging battery status. More tuning to be done, but this looks like it can potentially keep my project going for a couple of days with no sun.

Thanks again!

ralf-krause, 2023-08-24:
When do we get PiJuice with USB-C?

uGiFarukh, 2024-04-21:
Any update on this guys?