PiSupply/PiJuice #581: Firmware IDE files and build instructions

shawaj, 2020-11-12:
Add these to repo if possible

ryanteck, 2020-11-13:
This would be handy as could help with deciding alternative chips.

shawaj, 2020-11-13:
I have asked @mmilann for them so hopefully should have soon :-)

Think he is just moving it to the new IDE

shawaj, 2020-11-13:
Full set - [V1.4_2020_01_17-20201113T162716Z-001.zip](https://github.com/PiSupply/PiJuice/files/5538078/V1.4_2020_01_17-20201113T162716Z-001.zip)

IDE - [PiJuice_STM32CubeIDE.zip](https://github.com/PiSupply/PiJuice/files/5538082/PiJuice_STM32CubeIDE.zip)

Sources - [Sources.zip](https://github.com/PiSupply/PiJuice/files/5538083/Sources.zip)

@tvoverbeek @ryanteck what do you think is best way to store these in GitHub?

ryanteck, 2020-11-13:
Possibly a new github repo maybe and the put the source there. Releases then get tagged as releases.

The IDE Zip looks like the source code too/ the actual project from STM32 Cube.

shawaj, 2020-11-13:
@ryanteck we already have some source files here - https://github.com/PiSupply/PiJuice/tree/master/Firmware/Source

shawaj, 2021-03-17:
@mmilann do you have the updated files for these from v1.5 updates?