PiSupply/PiJuice #582: Using both charging sources.

fharryx, 2020-11-14:
What would happen if both charging sources would supply power?

shawaj, 2020-11-14:
You mean using a power supply in both the Pi micro USB and the PiJuice micro USB?

Or something else?

fharryx, 2020-11-14:
Yes, that is exactly what I mean!

tvoverbeek, 2020-11-14:
The strongest one wins. See this part of the schematic:
[image]
The CS1 and CS2 go to the microcontroller.

shawaj, 2020-11-15:
@fharryx as Ton has said.

It's ok to plug into both at the same time.

Some people plug one into mains and one into a solar panel for example.

Do you have any other questions or can we close the issue?

fharryx, 2020-11-15:
Thank you very much! My question is answered.

closed 2020-11-15 by fharryx

shawaj, 2020-11-15:
Perfect thanks very much 🤠