PiSupply/PiJuice #584: Test-Points for the I2C-2 bus

jophut, 2020-11-17:
Maybe this issue could be picked up in #580?

I'm using the PiJuice for a powering a Raspberry-Pi 3 in a portable device.  
To make the device more user friendly I hacked in an OLED display via the I2C-2 bus. Quite a bit of work to hack the firmware.
For the OLED I'm using the [U8G2 library from olikraus](https://github.com/olikraus/u8g2)

The most difficult hardware modification was getting the SDA and SCL lines out, most of the nets have a test-pad except the I2C-2 bus, I currently soldered some 36AWG enameled copper wire to PB10 and PB11 of the MCU.
[image]

So my question is: Would it be possible for future PCB's to have test-pads for the SDA and SCL line of the I2C-2 bus?
Also a test-pad for the SW1, SW2, SW3, IO1, IO2 would be really nice for me.
If there is any space left also for all three channels of both the LED's.

For my project it would be fine if they are placed at the battery area of the PCB because I use an external battery and thus that surface-area is free. This way it would be more easy to solder some hookup wire's.

Currently my modification is soldering the 36AWG wire to a [Molex Pico-EZmate 6 way connector](https://www.molex.com/molex/products/part-detail/pcb_headers/0781715006):
[image]
And then use superglue to fix the connector to a free pcb area.

Finally I use some [Molex Pico-EZmate Cable Assembly](https://www.molex.com/molex/products/part-detail/cable_assemblies/0369200603) to connect to external buttons and the OLED display.
[image]

shawaj, 2020-11-17:
@mmilann can we add this do you think? 

@jophut cool about the firmware hackery. Are you going to write up your modifications somewhere? Would be cool to see and share :-)

mmilann, 2020-11-17:
@shawaj as much as free space allows.

shawaj, 2020-11-17:
Great thanks very much @mmilann 👌

shawaj, 2020-12-05:
Closed this in favour of #580 and added comment there.

@jophut if you decide to write up anything about your firmware modifications or anything like that we would love to see it. Reply here if you do :-)

closed 2020-12-05 by shawaj