PiSupply/PiJuice #585: Solar Panel input showing as bad

thedataproject, 2020-11-18:
I have a the following setup:

- Raspberry Pi 3B+
- PiJuice HAT
- Adafruit USB / DC / Solar Lithium Ion/Polymer charger connected to J4 (https://core-electronics.com.au/usb-dc-solar-lithium-ion-polymer-charger-v2.html )
- Adafruit 6V 2W Solar panel (https://core-electronics.com.au/medium-6v-2w-solar-panel-2-0-watt.html)

Battery charges fine when connected to the power via the micro USB but when attempting to charge via the Solar Panel input shows as bad. 

Is there something i am doing wrong for this not to work?

shawaj, 2020-11-18:
@thedataproject you don't need the charger module in between. You can plug the solar panel directly into J4. This is likely to be what's causing the issue.

thedataproject, 2020-11-20:
Thanks @shawaj , i tried connected it directly to J4 but still shows as bad

shawaj, 2020-11-20:
@thedataproject in full sunlight? Sometimes it can show bad because the input is not high enough. 6v 2w is only 330mA - the input requires about 200mA. So I would say you will probably only get it showing "good" if there is full sun on the panel - or alternatively by getting a larger panel

ryanteck, 2020-11-25:
I would concur with the above, we'd typically recommend at least a 5W solar panel as for it to provide 2W would be in 100% perfect sunlight. It'd likely provide some charge when the Pi isn't turned on but with the Pi Turned on wouldn't be enough.

shawaj, 2020-12-05:
@thedataproject did you manage to test with a bigger panel at all? 

Can we close this issue now or do you need more assistance?

dingetje, 2021-05-18:
I have the official 12W panel and it showed as BAD/WEAK. Granted it was a bit cloudy, but at some point I had full sun on the panel and it still only showed WEAK and battery (5000 mA LIPO) percentage was slowly dropping despite being charged. So these panels are a bit of a joke for a Pi 3B. Disappointed.

tvoverbeek, 2021-05-20:
You cannot expect to run the Pi and charge the 5000 mAh battery at the same time with the 12W solar panel.
Your 12 W panel has 2 sections. Each one can only deliver 1A max at 5V (at full sun). Your 5000 mAh battery
uses a 0.925A charging current in the constant current charging phase.
I assume you only have 1 section connected to the PiJuice micro USB connectot.
The Pi3B uses about 500mA or more depending on what peripherals are attached.
This does not leave enough power to charge the battery, hence the BAD/WEAK status.
Try to connect both sections in parallel (needs soldering or special Y cable).

dingetje, 2021-05-20:
@tvoverbeek good to know, there was very little information provided with the panel. I did see 1A current in full sun, I'll create a Y cable and try again. Thanks.

leighghunt, 2023-11-09:
Found this ticket whilst looking for something else - but just to add some real world results here in case it helps anyone - I've got about a dozen remotely powered PiJuices with Pi Zeros and 15Ah LiFePO4 batteries and some of the official PiJuice 12Ah LiPo batteries, running 12 hours a day.

During winter here in New Zealand, some 40W panels have kept all the pis running most of the time with the exception of sustained bad weather. Pis were mounted to lamp posts and had good view of the sky. I also tested a 10W and 20W panel - the 10W panel was next to useless - the 20W panel would keep things going on sunny days, just about.

I had one Pi that was under partial tree cover that obscured about 50% of the sky, with panel at ground level - this one didn't manage to keep the battery charged even with 40W, so we swapped out for an 80W panel and this is now happy.

As we're enter summer, the cameras are generally staying above 80% charge all of the time - in the last couple of weeks, we had 2 rainy days in a row, and all seemed to stay above 30%.

Worth noting that the power draw of a Zero v1 W will be less than a Pi 3B+, but I am also intermittently running a cellular dongle - I have to power this up and down for 15 minutes between uploading photos as it draws too much power if left on all the time.

The panels I'm running are 12V ones, and I'm putting through [LM2596/XC4514 DC-DC Buck Converter](https://www.jaycar.co.nz/arduino-compatible-dc-voltage-regulator-module/p/XC4514)

More info if anyone is interested - https://github.com/venari/timelapse