PiSupply/PiJuice #586: Intermittent Communication Errors

nzerdh, 2020-11-18:
I'm having some issues with a PiJuice HAT that I've installed onto a Pi Zero W. I intermittently run into "Communication_Error" and have no idea how to fix it. I've already tried a different Pi Zero W but the problem still exists.

Sometimes the Pi boots and immediately cannot detect the hat. Rebooting sometimes helps. Sometimes the battery has to be disconnected so when the pi reboots it sees the hat.

Some times when it does connect, GPIO Power Input says it's "weak". Also, when it does connect, if I lose USB power, the Pi Zero W loses power... battery power does not kick in. 

What am I doing wrong?

nzerdh, 2020-11-25:
I purchases another PiJuice and it seems to work fine as is. There might be a problem with the first one. How do I get it resolved?

shawaj, 2020-11-25:
@nzerdh have you tried reloading the firmware to the PiJuice?

ryanteck, 2020-12-01:
Hi @nzerdh ,

To do this you would follow the instructions at https://github.com/PiSupply/PiJuice/tree/master/Firmware , replacing the end of the command shown with /usr/share/pijuice/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary

So running ```pijuiceboot 14 /usr/share/pijuice/data/firmware/PiJuice-V1.4_2020_01_17.elf.binary``` should do the job.

nzerdh, 2020-12-07:
sorry about the delay. Was finally able to try reloading the firmware. After sending the command suggested above, there was immediately a communication error.

Once I power-cycled the whole thing and got communication back, I tried reloading the firmware with bootloader mode. rebooted the Pi and it seemed to work ok. Could run it with the battery alone. Or USB only. Could use power it either USB on the Pi or on the PiJuice Hat.

However, once I clicked on "Reset to default configuration" via the PiJuice HAT Configuration gui, I received the communication error again HAT Leds turn off. Reboot does not restore communication. Must shut down completely. Take out the battery, Remove USB power. Upon restoring power, battery inserted, pi boot to gui, battery is not recognized -- must remove battery and reinsert to get recognized.

When I go to "Wakeup Alarm" and click on "Set RTC Time"... PiJuice Hat LEDs turn off, GUI says there's a communication error.

any suggestions?

ryanteck, 2020-12-10:
@tvoverbeek do you have any suggestions for this?

tvoverbeek, 2020-12-10:
It smells like fluctuating power. Any hungry peripherals attached to the Zero W?

An other possibility is a fluctuating clock due to varying cpu load.
You could try to add the line `force_turbo=1` to /boot/config.txt and reboot.
This will keep the clock constant.

nzerdh, 2020-12-12:
No peripherals. Tried two different Zero Ws. Fresh OS installs.

problem persists even after the "force_turbo" addition to config.txt

nzerdh, 2020-12-13:
Yes, still the first PiJuice HAT. (2nd one is still ok).

Here's a screenshot of the i2cdetect command with the PiJuice GUI windows open. What seems weird to me is that between yesterday and today, that battery went down to 0% without the Pi running. It is charging while I'm typing (3% now).

[image]

Of course, when I get the "communication error" i2cdetect shows only dashes.
[image]

tvoverbeek, 2020-12-13:
On the first screenshot GPIO Power Input is shown as WEAK.
This implies your power supply is connected to the RPi and does not have sufficient power to both power the RPi and charge the battery and power the PiJuice properly.
What does GPI Power Input show with the working PiJuice?
Also the i2cdetect output shows 68 and not 'UU' although the eeprom address is set at 0x50.
It should have loaded the RTC driver automatically (unless you prevent it in /boot/config.txt).
Does your working PiJuice show 'UU' in the i2cdetect output for address 0x68?

nzerdh, 2020-12-14:
Sorry, I'm pretty new to this. I'm not understanding what exactly I should be seeing.

I'm using a 10W USB power supply. How much does it need?
Along with "WEAK" sometimes I get "BAD".
[image]

I don't think I'm blocking the RTC driver with the /boot/config.txt, the only thing I've modified in that txt is the line "force_turbo=1" recommended earlier.

The 2nd PiJuice (that is working) was installed with a Pi 4. I don't see "UU", here's what it displays when I run i2cdetect -y 1
[image]

nzerdh, 2020-12-14:
I also have another Pi Zero with PiJuice Zero that's working as expected. Here's its "i2cdetect -y 1" output.

[image]