PiSupply/PiJuice #587: Deep sleep for a long time

FNumidia, 2020-11-18:
The PiJuice MCU has to remain active (albeit in low power mode = the dim green blinking) to be able to react on events like 'Charging restart` and '(repeated) alarm wakeups.
If the power completely goes off the firmware loses the RTC and any volatile data.
(Same answer as @shawaj answer above)

_Originally posted by @tvoverbeek in https://github.com/PiSupply/PiJuice/issues/569#issuecomment-715907892_

In the case of a very long period of deep sleep (5 months or more) after which the battery is completely empty we will have the same effect of a firmware reset? In this case could the RaspberryPi wake up it automatically if the power is on again (supposing to have previously set Wake Up on Charge > 10%) or I would need to push the button? Would the battery (BP7X) charge again or it would be seriously damaged?

tvoverbeek, 2020-11-19:
Since the Wakeup on charge setting is not saved in the nonv volatile memory, I am afraid you need to push the button.
Of course if your external power is connected to the RPi, the RPi will boot when power returns.
The PiJuice will start will default settings (as after a firmware reset).
I do not believe the battery will be damaged, since the PiJuice will stop when the battery voltage goes below a safety limit.

shawaj, 2020-11-19:
@tvoverbeek do you think this is something we should move to non volatile memory? Or at least provide the option to save it there? As part of #580 perhaps

tvoverbeek, 2020-11-19:
It was once in non volatile memory, but @mmilann removed it very early on.
I guess to avoid boot loops if the Pi gets woken up when the battery level is too low, so it immediately shuts down again.
If we put it back in non volatile memory, then it has to also have an enable/disable option in non-volatile memory.
Check with @mmilann what the reason was to not include it in non-volatile memory.

mmilann, 2020-11-20:
This will be revised in next update.

shawaj, 2020-12-24:
See #612