PiSupply/PiJuice #588: PJLIPO_12000 Current Charge not Correct

sethwebster, 2020-11-19:
I purchased the PJLIPO_12000 and have it installed and generally working. I've set a recurring alarm that wakes it, does some work, and shuts down the Pi. All of the above is working.

I have a message sent out when it boots reporting the current battery capacity via ```pj.status.GetChargeLevel()``` (which normally works but intermittently fails with `COMMUNICATION_ERROR`). The capacity hovers around ~70% - 80%. At some point the battery is dead and recharges fine. The reported battery capacity upon plugging back in to charge was 90%+.

I have configured the settings using the provided profile.
[image]

Incidentally, another issue is that I cannot use the `NTC` temperature sensor as it always reports 240˚ and suspends charging; I have it set to `On Board` which seems to work and the temperature does seem to move with charging, etc.

Any suggestions on how to get an accurate battery level reading?

TIA!

shawaj, 2020-11-19:
think this might relate to #577 - at least the NTC part...

sethwebster, 2020-11-19:
¯\_(ツ)_/¯ I didn't see anything in there that helps but I'll try reconnecting everything. I've tried a number of the troubleshooting steps in there.

Thanks!

sethwebster, 2020-11-25:
Any luck?

sethwebster, 2020-12-01:
I've still been unable to sort this. I've tried a different battery, a different Pi Zero... Any suggestions would be appreciated. FWIW, I also noted that I experience the charging temp issue from #577 as well.

tvoverbeek, 2020-12-02:
@sethwebster Which RPi are you using? It maybe that changing clock frequency is causing problems with I2C communication.

sethwebster, 2020-12-02:
Pi Zero W

tvoverbeek, 2020-12-02:
@sethwebster To force a constant clock frequency (and a stable I2C clock) add the line `force_turbo=1` to /boot/config.txt and reboot. See if this helps.

sethwebster, 2020-12-04:
Did so. Currently running it all the way down and having it log the battery % along the way every 30 mins.

sethwebster, 2020-12-04:
Well, it's down to 65% which I've never seen before. Never saw lower than 84% previously so tracking in the right direction. Still seeing the temperature warning when using the NTC option instead of ON_BOARD.

sethwebster, 2020-12-10:
¯\_(ツ)_/¯ It seemed better, then I am experiencing something new --perhaps related?

What I am trying to do is the often used power-up, do some work, set a wakeup, power down. Wake-up, repeat.

I haven't been able to figure out whether the battery level test is working b/c after a day or so (sometimes 2, 3), the PiJuice "goes offline". I can't get it to power back on unless I unscrew the power (red) terminal or the wired battery. Once disconnected, the status light goes Red. I then reconnect the power and the light goes blue. Then, all is well for a while. Repeats after day or two. Doesn't seem to matter whether power is connected to the power-in or not.

I'm stumped.

tvoverbeek, 2020-12-10:
In your sequence how do you do the last part; set a wakeup and power down?
I assume with some python code.
I also assume your Zero W has an internet connection via its wireless interface. Correct?
It could be the wakeup ebabled flag gets cleared, then the next scheduled wake up will not work.

sethwebster, 2020-12-10:
Yes, Python. Yes, internet.

I cannot power it back on via the buttons. I have to physically disconnect the power from the battery. This doesn't seem wakeup flag related, does it?

Thanks for all the help.

tvoverbeek, 2020-12-12:
I guess you know that when you have an internet connection the systemd-timesyncd daemon will sync the time to the system clock and copy it to the RTC. This will also clear the rtc wakeup flag. If you do not set it yourself again at shutdown the next wakeup will not work.
This is only an issue with the PiJuice HAT if the ID-EEPROM address is left at its default 0x50. Then the device-tree blob in the HAT eeprom will cause the kernel driver for the RTC to be loaded.

Something else, you wrote it occurs after 2,3 days the problem occurs. Somewhere a memory leak? The ZeroW has 'only' 500Mb memory.

sethwebster, 2020-12-19:
Memory-leak-wise, I don't think so. It shuts down fully between runs and starts anew. It's script that downloads a pdf and renders it to an eink display, then shuts down. Yeah, I re-set the time each boot and that is working (I figured this out early).

sethwebster, 2020-12-19:
I guess I still don't understand how this might be a software problem when I have to physically disconnect a wire to get it to come back.

Also, the force_turbo thing seems to help when I leave it running continuously, but not in this power-up, work, power-down scenario.

Thanks for all the assistance.

sethwebster, 2021-01-04:
An update. It did that thing again where I'd need to manually disconnect. Instead of doing so, I powered the pi directly, and fired up the pijuice_cli. 

This is the status page:

[image]

sethwebster, 2021-04-05:
Bumping this one in case anyone has ideas :)

tvoverbeek, 2021-04-06:
Which question? The incorrect charge level or the not waking up?
Have you updated to the last version of the software (1.7) and firmware (1.5).
Do the problem(s) still occur with the updated software/firmware?