PiSupply/PiJuice #590: Potential for RTOS

mmilann, 2020-11-19:
FreeRTOS is integrated in stm32cube SDK and could be pretty straightforward to introduce. 

Not sure for mbed. Logging and debugging over cli service can be beneficial for PiJuice. Current RAM usage is 22K/32K and Flash ~80K/256K.

@tvoverbeek @ryanteck @shawaj any thoughts?

ryanteck, 2020-11-19:
Potentially if possible using MBED could be good as then it means we have a wider range of alternative ICs for #579 that we could consider without having to worry about software as being as much of an issue.

shawaj, 2020-11-19:
@mmilann yeah it sounds good to and I agree with @ryanteck with regard to the alternative ICs etc

mmilann, 2020-11-19:
@ryanteck @shawaj I am concerned about ease of porting current drivers into mbed, if it would not be a big task, like need to rewrite drivers contained in cube sdk then it is worth going that path and continuously add features.