PiSupply/PiJuice #591: Charging problem with 3A Supply on Raspi 3B

MCKly, 2020-11-19:
Hi PiJuice - Community!

I have the following problem:

I bought a PiJuice for my Raspi to bridge some power loss that sometimes occur on my power line (like a USV). After 13 month now the provided battery (BP7X, produced 2019-03-15) thickened and was no longer usable. Within this I recognized that the power supply shows a malfunction and I replaced it with a new power supply with 3A. After reconnecting everything, and installing a new battery (BP7X), the status of the GPIO power input changed from "present" to "not present", and the battery was not charged. Only if the power regulator is set to "LDO", the status of the power supply changed to "bad". After this I tried a series of other power supplies and found one that work. But surprisingly, it was one with only 1A. Then I thougth that the new 3A power supply is maybe broken, so I bought a second one, with the same result. The power supply that worked before the whole year had only 1.5A, too.

Here are some more details to the used hardware/software:

Pijuice: V2.2.2, Firmware 1.4
Raspberry Pi 3 Model B Rev 1.2, Raspbian 10, Kernel 4.19.75-v7+

I have now really the problem that I want to use a power supply with enough power for the raspi and it's attached PiJuice. Have someone an idea what could be the problem? Many tanks!

tvoverbeek, 2020-11-19:
Is this with an official RPi power supply or with something like a phone charger?
Some power supplies can have a large ripple on their DC output which causes all kind of problems.

MCKly, 2020-11-22:
Thank you very much for your answer! I will check this issue. I bought it from a local quiet “big” it-company, which tells that the power supply is “suitable for Raspberry Pi 3. I will order an original one to see what happens. I will inform you when I have the results, hopefully on the next weekend.
[Only to complete this: I have bought this power supply (German page/language only....):
https://www.conrad.de/de/p/voltcraft-sps-3000-r-steckernetzteil-festspannung-5-v-dc-3000-ma-15-w-raspberry-pi-3-geeignet-1594595.html
]

tvoverbeek, 2020-11-22:
@MCKly Does your 3A power supply work fine with only the Pi3?
I could not find anything about ripple in the Conrad specification (I am in Austria, no problem with reading German).
The Pi3 alone uses far less than 3A, so there will be less ripple then.
So it is probably 'Raspberry pi 3 geeignet'.

ryanteck, 2020-12-02:
Hi @MCKly ,

Did you check with a different power supply?

MCKly, 2020-12-02:
Sorry for the late answer - I'm still waiting for one of the "original Raspberry" one, to test it. The delivery time is at the moment quite high... Hopefully I got it on Friday to test it. My power supplies from "Conrad" are working fine with the Pi alone.

MCKly, 2020-12-08:
Fortunately, I got the original Raspi Power supply (micro USB variant). Now it seems everything to work...
**Thank you very much for your help!**
I really would not have so fast the idea that a “for raspberry suitable power supply” is not working with the PiJuice.

tvoverbeek, 2020-12-08:
Glad it works now. Closing the issue. If you do not agree please reopen it or start a new issue if it is a different problem.

closed 2020-12-08 by tvoverbeek