PiSupply/PiJuice #592: raspi4 pijuice and poE installed

envxcom, 2020-11-23:
Hello
For a projet i have intalled on a board Hat three cards
Raspi4
loragateway
PiJuice HAT power
and poE switch ethernet

when i connect the battery everything turns on raspi and lorawan card the ethernet cable are connected between poe and raspi.

the battery is charged I unplug the power cable but the battery drains quickly and the power does not go through the ethernet cable connected between the raspi and the poE

an idea for help me
Thank you

Roland

shawaj, 2020-11-23:
@envxcom can you tell me which PoE board you are using? Can you post a link?

And how are you connecting all the boards together?

nik-1337, 2020-11-24:
Hi, 

Did the same thing today with the same result. I am using this board https://www.amazon.de/gp/product/

any ideas is there a manual switchback ?

shawaj, 2020-11-24:
@nik-1337 your link doesn't go anywhere?

nik-1337, 2020-11-24:
sorry 

https://www.amazon.de/gp/product/B07JPXR9ZN

shawaj, 2020-11-24:
@nik-1337 so you have that board and then the PiJuice on top? And what happens?

What PoE injector are you using?

nik-1337, 2020-11-24:
different POE Sources

If i Power Up the PI original vom the POE Power Source its laoads the battery if i disconnect the cable the battery dosent load load anymore

shawaj, 2020-11-25:
@nik-1337 what do you mean by different PoE sources? It is an 803.2af injector right?

If you do not have the cable plugged in then the PoE will not provide any power.

I am not sure I am understanding your meaning.

nik-1337, 2020-11-25:
Af to Bt... 

Well.... 

If I do a fresh cold Start with Poe all works finde... Battery is charching and so on. But after an un and Re pluging of the rj4t the PiJuice dont change back from battery to Normal Poe

ryanteck, 2020-11-25:
@nik-1337 It could be that the POE module you're using on the Pi is faulty. Without the Pi Juice on top and the Pi Powered via POE could you check the voltage at the 5V Pins with a Multimeter? 

Having a look at the amazon reviews one mentions they measured a 1V swing in voltage, it could be if it's not adequately able to power the Pi the Pi Juice won't revert back if it doesn't see a stable input.

shawaj, 2020-11-26:
@envxcom do you have a USB-C power supply you can test with? Plug it into the Pi 4, then unplug and replug it and is the PiJuice charging?

ryanteck, 2020-12-02:
Closed as no response for 7 days, you can re-open if you still have an issue.

closed 2020-12-02 by ryanteck