PiSupply/PiJuice #594: Pijuice Communication Errors

nchase0514, 2020-11-25:
I currently have a PiJuice stacked on a Rpi 4.
Firmware is 1.4
Software is 1.6

I am getting communication errors when making changes in the GUI. It seems it happens if I make more than one setting change at a time before clicking the apply button. For example if I change the function of all three switches before applying the changes it causes communication error messages.

I also am unable to get LED 2 to work at all. It doesn’t matter which color I choose, nothing. Not sure if this is related or if the LED has failed.

It also seems button presses work intermittently. I have tested with user functions and all three switches work, but if I change the settings to perform functions like reboot or power off, nothing happens.

I have checked and the pijuice service is running even after the communication errors. Most of the time if I close the GUI and reopen the errors go away but the persistent issue is the buttons not working. I’m curious if there is a way to factory reset the PiJuice to clear out any changes I have made other than restoring default settings which I have tried and had no effect.

I would really appreciate any help. Overall the basic UPS function is working, which is great, but the buttons not working is a big frustration.

Thank you!

ryanteck, 2020-11-25:
@tvoverbeek Do you think this could possibly be a firmware issue and re-flashing it could fix it?

tvoverbeek, 2020-11-25:
@ryanteck I do not know, but I would first try a "reset to default configuration" either via the GUI (HAT configuration, General) or by a dual long press of SW1 and SW2 for 20 seconds (https://github.com/PiSupply/PiJuice/tree/master/Software#buttons).
During the reset both LEDs on the HAT should be flashing red/blue. This could prove if LED 2 is working.
If that does not work reflashing the firmware is worth a try.

nchase0514, 2020-11-25:
Hi,
Thank you for the quick response! I would like to try to reflash the firmware. I have tried resetting to default settings using both methods and I am still having communication issues and LED 2 still does not work. Is there instructions on how to reflash the firmware somewhere?

Thank you!

nchase0514, 2020-11-25:
@tvoverbeek I found the steps to follow in issue #164. My only question is which firmware version I should use?

shawaj, 2020-11-26:
@nchase0514 it's best to use the latest v1.4 firmware

nchase0514, 2020-11-26:
@shawaj thank you! I will give it a shot and see what happens. Thank you all for the support!

nchase0514, 2020-11-26:
Okay, I have followed the steps to get the PiJuice into boot load.
-Remove all power
-Hold Switch 3 while powering up Rpi
-Stop PijJuice Service
-Clone PiJuice github
-cd into PiJuice Firmware
-Attempt to run firmware binary file and am stuck (see below).

[image][image]

ryanteck, 2020-11-26:
> <https://user-images.githubusercontent.com/74981439/100295430-d1435880-2f46-11eb-9243-3d6f52d2a4bd.jpeg>[image:
> image]

I think it's missing the number 14 after the command but before the
firmware filename.

nchase0514, 2020-11-26:
I was able to successfully flash the firmware but upon reboot and starting pijuice service same communication issues as before and no LED 2.[image][image]

nchase0514, 2020-11-26:
Just to try something I uninstalled and reinstalled the pijuice software. Now the pijuice service won't start.

[image]

nchase0514, 2020-11-26:
Finally got service to start. No change, still getting communication errors. After removing software and reinstalling and flashing firmware, my original settings still remain, is this normal?

shawaj, 2020-11-26:
@nchase0514 what OS are you using? Is it up to date?

When you flashed the firmware what did you see when flashing it - a success message?

nchase0514, 2020-11-26:
@shawaj I am running Raspberry Pi OS and yes it is up to date. Just for fun I updated yesterday. Also, I did get a success message when flashing the firmware.

tvoverbeek, 2020-11-26:
@nchase0514 OK, a few steps back. Let us see if the Hat is detected properly.
Could you list the output of `i2cdetect -y 1`?
If i2cdetect cannot be found install i2c-tools (`sudo apt install i2c-tools`)
Also is i2c enabled in /boot/config.txt?

nchase0514, 2020-11-26:
The PiJuice is being detected[image]

tvoverbeek, 2020-11-26:
OK, so it seems basic I2C communication is working.
Is the ID-EEPROM address set t0 0x52? This because the RTC shows uo as 0x68 in the i2cdetect output.
If it was at the default 0x50 a RTC driver should have been loaded and the i2cdetect output for addr 0x68 then shows up as 'UU' meaning the kernel is using this address.
The output from `lsmod` would be useful to see which modules/drivers are loaded.
Are you running anything else on the Pi4? Maybe the varying clock frequency (due to varying CPU load) is causing the communication errors.

nchase0514, 2020-11-26:
The ID-EEPROM is set to the default 50. But I do not see UU at 68 in i2cdetect. 

The only thing I have running on the Rpi is a node red program that will eventually be my aquarium controller. Right now it only controls GPIO outputs for relays, a temperature probe and PWM out to control a fan that cools the Rpi.

Below is the output of lsmod:

pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
cmac                   16384  1
rfcomm                 49152  4
bnep                   20480  2
hci_uart               40960  1
btbcm                  16384  1 hci_uart
bluetooth             372736  29 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           16384  2 bluetooth
ecc                    40960  1 ecdh_generic
fuse                  118784  3
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
joydev                 20480  0
brcmfmac              319488  0
brcmutil               20480  1 brcmfmac
sha256_generic         16384  0
libsha256              20480  1 sha256_generic
rpivid_mem             16384  0
vc4                   241664  7
bcm2835_codec          36864  0
v4l2_mem2mem           32768  1 bcm2835_codec
bcm2835_v4l2           49152  0
cec                    49152  1 vc4
v3d                    69632  5
cfg80211              692224  1 brcmfmac
drm_kms_helper        184320  3 vc4
gpu_sched              36864  1 v3d
bcm2835_isp            32768  0
videobuf2_vmalloc      16384  1 bcm2835_v4l2
bcm2835_mmal_vchiq     28672  3 bcm2835_isp,bcm2835_codec,bcm2835_v4l2
rfkill                 28672  6 bluetooth,cfg80211
videobuf2_dma_contig    20480  2 bcm2835_isp,bcm2835_codec
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
drm                   466944  14 v3d,vc4,gpu_sched,drm_kms_helper
raspberrypi_hwmon      16384  0
videobuf2_v4l2         28672  4 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       57344  5 bcm2835_isp,bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
drm_panel_orientation_quirks    16384  1 drm
i2c_bcm2835            16384  7
pwm_bcm2835            16384  0
snd_soc_core          200704  1 vc4
snd_compress           20480  1 snd_soc_core
sg                     28672  0
snd_pcm_dmaengine      16384  1 snd_soc_core
videodev              237568  6 bcm2835_isp,bcm2835_codec,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_bcm2835            28672  3
snd_pcm                98304  4 vc4,snd_pcm_dmaengine,snd_bcm2835,snd_soc_core
vc_sm_cma              32768  7 bcm2835_isp,bcm2835_mmal_vchiq
mc                     40960  6 bcm2835_isp,bcm2835_codec,videobuf2_common,videodev,v4l2_mem2mem,videobuf2_v4l2
snd_timer              32768  1 snd_pcm
w1_gpio                16384  0
snd                    73728  11 snd_compress,snd_timer,snd_bcm2835,snd_soc_core,snd_pcm
wire                   40960  1 w1_gpio
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
cn                     16384  1 wire
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
i2c_dev                20480  14
ip_tables              28672  0
x_tables               32768  1 ip_tables
ipv6                  466944  68
pi@raspberrypi:~ $

shawaj, 2020-11-26:
@nchase0514 where are the other accessories you mentioned connected to? Are any of them on the I2C bus as well? Are they connected via the GPIO pins on the PiJuice?

ryanteck, 2020-11-26:
At the same time @nchase0514 could you take a picture of the underside of the board zoomed in primarily to the biggest chip on the back for us to have a look at.

Thanks

tvoverbeek, 2020-11-26:
I guess your node red program is mostly cyclicly polling resulting in the arm clock varying constantly.
Could you try the following:
in /boot/config.txt add the lines:
```
core_freq=500
core_freq_min=500
```
This will fix the clock and (after a reboot) see if you still get communication errors.

nchase0514, 2020-11-26:
@shawaj Currently I do not have anything connected besides the cooling fan which is connected to the Pijuice on pins:

4- 5v
6- GND
8- PWM

@ryanteck 
Here is a picture of the bottom side of the board
[image]

@tvoverbeek 
I have made the changes you suggested in boot/config.txt
So far no communication issues and the buttons are responding. I'll give it a bit and see if that solves it. Thank you!

nchase0514, 2020-11-26:
@ryanteck @tvoverbeek @shawaj Okay, seems as though the issue is fixed. No more communication errors, and buttons are now responding. Thank you all for the help. I absolutely love the PiJuice and am so happy it is running smoothly again. Thank you very much!

shawaj, 2020-11-26:
Great thanks @nchase0514 for letting us know

closed 2020-11-26 by shawaj

ryanteck, 2020-11-26:
Great to hear! I'll close your email ticket too.

nchase0514, 2020-11-26:
@ryanteck Thank you very much!