PiSupply/PiJuice #595: PiJuice Communication Errors

thedataproject, 2020-11-26:
Hi,

I have a PiJuice attached to my Pi 3B+ which is working perfectly (firmware v1.4). However when i attach a DFRobot Gravity I2C ADS1115 16-Bit ADC Module (https://wiki.dfrobot.com/Gravity__I2C_ADS1115_16-Bit_ADC_Module_Arduino_&_Raspberry_Pi_Compatible__SKU__DFR0553) the pijuice shows communication errors.

It is attached like below (with the exception of being attached to the Pijuice not the Pi.

[image]

shawaj, 2020-11-26:
@thedataproject it looks like that module can use I2C addresses 0x48 or 0x49 neither of which conflict with the PiJuice. But just for completeness have you tried switching the address change switch?

Also, are you sure you haven't mixed up the SDA and SCL lines connecting to the I2C bus?

Lastly, what output do you get from `sudo i2cdetect -y 1`?

thedataproject, 2020-11-26:
With just the Pijuice 
[image]

When the ADC is plugged in
[image]

ryanteck, 2020-11-26:
@thedataproject It looks like you're powering the I2C Module with 5V if wired as per your fritzing schematic.

I suspect this is causing the voltage to be too high for the Pi to pull low possibly.

Could you try powering it via the 3V3 pin instead?

tvoverbeek, 2020-11-26:
@ryanteck Yes, you are probably right. The ADS115 module has 2 10K pull-up resistors on SCL and SDA connected to VCC (here 5V). The module supports 3V3 according to the documentation.

thedataproject, 2020-12-01:
I have switched it to the 3V3 as suggested and i am still getting the communication error for the PiJuice and cannot access the ADS115 module (raise IOError(ffi.errno IOError: 5)

ryanteck, 2020-12-01:
Hi @thedataproject ,

Sorry it seems I might have been incorrect and that the module might need to be powered via 5V on the Pi. It seems they have a regulator on it to drop the voltage down to 3v3 on the i2c lines.

The issue however is if the Pi Juice is communicating without the ADC plugged in that would confirm there's not an issue with the Pi Juice and it's a setup issue. 

The only thing I can think in that case is possibly the 10k pullups on the ADC are causing issues, altough typically they shouldn't interfere with the Raspberry Pi's all I can think is the Pi Juice skews it enough that the not normally required pullups on the ADC are interfering.