PiSupply/PiJuice #597: SYS_FUNC_HALT_POW_OFF - decreasing 60s timeout

kubiaj, 2020-11-27:
Hi,
is there a way how to decrease 60s timeout in SYS_FUNC_HALT_POW_OFF function? I would like to power off the RPi as soon as possible (let say <10sec timeout).  RPi is enclosed in a box so I see only few LEDs and I dont know for a 60sec if it will turn off or not.

tvoverbeek, 2020-11-27:
Modify the timeout in /usr/bin/pijuice_sys.py line 61:
https://github.com/PiSupply/PiJuice/blob/a050850c1295d2c34198f1217103c94a7a44dce9/Software/Source/src/pijuice_sys.py#L61
but make it long enough so the Pi can shutdown properly before the power disappears.

ryanteck, 2020-12-02:
Hi @kubiaj , did the resolution tvo propose resolve your issue?

kubiaj, 2020-12-02:
Hi, 
I did some preliminary test with 10s timeout yesterday and it turns out that it have some issues and it didn't power off the RPi at all. RPI shut down itself correctly after holding the SW1 pin for a cca 10 sec but the pijuice didn't cut off the power even after hours.   I need to retest it with default 60sec and then reduce it again.  Is there any known issues?

Im using RPi4 with the latest Raspbian (fully updated).

tvoverbeek, 2020-12-02:
@kubiaj How are you powering the Pi4/PiJuice combination? The power off only works when external power is connected to the PiJuice or if there is no external power (RPi running on battery).
Are you sure the 10 seconds are long enough? Have you seen the 10 flashes from the green RPi led indicating end of shutdown?

kubiaj, 2020-12-02:
Hi
I just tested and it even with 60sec timeout and it is also not working.  Maybe I did something wrong ... 

What I need it to use RPI + Pijuice with 12Ah battery as a primary source during the demo, during this time it is unplugged from USB/J4 power source.  So I need a way how to power it up (SW1 single press: HARD_FUNC_POWER_ON) and power it down (SW1 LONG_PRES1: SYS_FUNC_SYS_OFF_HALT).

Charging/powering is done via J4 (POW) (actually it is a USB cable power lines soldered to this input, I don't have enough space to fit in the USB cable ).

During my tests it is also powered via the J4 (POW). Do you see this as an issue?

kubiaj, 2020-12-02:
Re  Have you seen the 10 flashes from the green RPi led indicating end of shutdown?)

I dont see this, I only see that after LONG_PRESS1 the D2 do some funny blinking and then the RPi start shut down.

kubiaj, 2020-12-02:
Is there a way a to reset all configuration into the default state.  I can power it off now, unless I hold the SW1 for a 20 sec.

tvoverbeek, 2020-12-02:
@kubiaj You wrote:
```
So I need a way how to power it up (SW1 single press: HARD_FUNC_POWER_ON) and
power it down (SW1 LONG_PRES1: SYS_FUNC_SYS_OFF_HALT).
```
SYS_FUNC_SYS_OFF_HALT is the wrong function. It turns off VSYS not the 5V to the Pi.
You need SYS_FUNC_HALT_POW_OFF connected to SW1 LONG_PRESS1 (as in the issue title).

kubiaj, 2020-12-02:
@tvoverbeek  Argh, I missed this. Now it is working fine with 15sec timeout - many thanks. 

I have one additional question.  Im using RPi4 + PiJuice HAT with external battery  (12AH Pi Juice)+ external button and Im powering it via J4 or USB + Im using the RTC  + Im powering external device via Rpi4 USB ports (cca 0.3A).   Could I use PiJuice Zero instead of the HAT? It is smaller and it is not sold with battery onboard which I don't want to have.

tvoverbeek, 2020-12-02:
@kubiaj Yes, you could use the PiJuice Zero. It does not have the EEPROM (it is not a HAT), so you need to load the RTC driver with a dtoverlay statement in /boot/config.txt (only if you need OS support for the RTC).
Otherwise it has the same circuitry as the HAT (minus one LED). See the hardware section on Github for some other small differences.

kubiaj, 2020-12-03:
@tvoverbeek Perfect thanks.

closed 2020-12-03 by kubiaj