PiSupply/PiJuice #598: PiJuice Service failed to communicate with PiJuice service

tmourik, 2020-11-30:
Good afternoon,

I have a problem with the PiJuice HAT that I use on the Raspberry Pi 3B+. Attached to the PiJuice HAT is a 12.000MaH battery that is specially for the PiJuice HAT.

When I want to make an adjustment in the configuration I get the message "Failed to communicate with PiJuice service". (See picture for error message). Unfortunately I could not find a solution on the internet. I hope someone can help me with this.

Kind regards,

Timo van Mourik
[image]

tvoverbeek, 2020-11-30:
You get this message when the service is not running or when the file /tmp/pijuice_sys.pid does not exist or cannot be read.
What is the output from `ls -l /tmp/pijuice_sys.pid`?
It should look like this;
`-rw-r--r-- 1 pijuice pijuice 3 Nov 28 20:58 /tmp/pijuice_sys.pid`
What does `systemctl status pijuice.service` say?
If the service is running a sample output is:
```
● pijuice.service - PiJuice status service
   Loaded: loaded (/lib/systemd/system/pijuice.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-11-28 20:58:00 GMT; 1 day 17h ago
 Main PID: 584 (pijuice_sys.py)
    Tasks: 1 (limit: 1939)
   CGroup: /system.slice/pijuice.service
           └─584 /usr/bin/python3 /usr/bin/pijuice_sys.py

Nov 28 20:58:00 rpi-pj-mch systemd[1]: Started PiJuice status service.
```

tmourik, 2020-12-01:
@tvoverbeek After 2 reboots "systemctl status pijuice.service" said the service was active. I didn't even change anything!
But every time i reboot the system still doesn't recognize the UTC time.  Is there a solution for? I thought it would be fixed when the pijuice.service was running again.
[image]

ryanteck, 2020-12-01:
Hi @tmourik , could you let me know what you get when you run ```sudo i2cdetect -y 1``` in the console?

tmourik, 2020-12-01:
Hi @ryanteck 

This is what i get:

[image]

ryanteck, 2020-12-01:
Thanks,

It seems that it is not reading the eeprom for some reason and loading the RTC overlay.

Could you try adding ```dtoverlay=i2c-rtc,ds1339``` to the bottom of your /boot/config.txt file. If this resolves it we can then work from that.

tmourik, 2020-12-02:
Hi @ryanteck
 
I added 'dtoverlay=i2c-rtc,ds1339' where you asked for. As you see after a reboot the systems sets the UTC Time correct.

Thank you for your help.
[image]
[image]

closed 2020-12-02 by tmourik

ryanteck, 2020-12-02:
Great to hear!

tvoverbeek, 2020-12-02:
@tmourik If this works then I guess your EEPROM is set to 0x52 or you have disabled reading HAT EEPROMs.
Also the automatic setting of the RTC only works when you have an internet connection. The systemd-timesyncd daemon will set the system time from the internet and copy it (in UTC) to the RTC.
Note that this may cause problems when using alarms in the PiJuice GUI (Every time timesyncd updates the time the wakeup enabled is reset).

tmourik, 2020-12-03:
@tvoverbeek thank you for the extra information.