PiSupply/PiJuice #599: Authentication errors

EnWrapTured, 2020-11-30:
Hi

I have installed a PiJuice on raspbian on a pi 3 model b rev 1.2

If I try to change many of the setting in PiJuice_cli I get: "Failed to communicate with PiJuice service" (e.g. turn on watchdog timer)

If I run systemctl status it says the service is dead.

If I try to restart it it says authentication failure

I am logged in as pi

There are also other odd things, like if I set up a script for power loss it does not run.

The UPS part of the PiJuice works perfectly I just can't seem to control it. I installed using apt-get

Help?

tvoverbeek, 2020-11-30:
See also #598 
You can try to run the service manually in a terminal window `sudo -u pijuice python3 /usr/bin/pijuice_sys.py` and see if you get an error message.

EnWrapTured, 2020-11-30:
thanks very much, it is working now!

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

closed 2020-11-30 by shawaj