PiSupply/PiJuice #600: Battery not present in pijuice_cli, bad connection?

hlarnier, 2020-12-02:
Hi,

I just got a PiJuice for some autonomous projects. I removed the plastic tab, and everything looked fine for the first day, but now it seems like the battery is not detected when mounted on the board. Battery is a BP7X shipped with the PiJuice.
Here is a screenshot of pijuice_cli:
[image]

By wiggling the battery a bit (or just randomly), I can have it display present and at 98%, so it seems unlikely to me that it's an issue with the battery itself. Charging is also enabled if that helps. 

Hardware:
Raspberry Pi 4B
PiJuice, Firmware 1.4
Battery: BP7X

Any ideas of what I could try out to solve this issue or do a proper diagnostic?
Thanks

tvoverbeek, 2020-12-02:
Smells like bad contacts between the battery and the PiJuice.
Can you adjust the plastic tabs which hold the bP7X so there is more pressure on the contacts?
Maybe that helps.

hlarnier, 2020-12-02:
Thanks for the quick reply.
That's kind of what I thought of as well, but even if I carefully align the pins, the LED1 sometimes flashes for a brief moment indicating contact and reverts back to red, and pijuice_cli shows that brief contact but goes back to NOT PRESENT.

Anything I can test to see if it's an issue with the board or the battery itself?

Knusk, 2020-12-28:
Same issue for me, with the same hardware. I haven't tried wiggling, though ... There are seemingly a lot of options in the preferences, so I can not guarantee that I have done everything correctly there.

kaulex99, 2020-12-31:
Have had the same problem, a new battery solved it.

hlarnier, 2021-01-12:
Ok, thanks all, I can confirm it as well.