PiSupply/PiJuice #602: PiJuice  3D models

kubiaj, 2020-12-03:
Hi HW developers,
could you release a 3D models for PiJuice + PiJuice Zero or release some documentation with dimensions? As you are probably already using Altium designer it should be very easy to generate a 3D model in STEP format. 

It would be very helpful to have this as Im trying to do custom modification  of some case and without the dimensions I don't know where to create holes for LEDs and etc.

shawaj, 2020-12-03:
See here https://github.com/PiSupply/PiJuice/tree/master/Hardware/3D

closed 2020-12-03 by shawaj

hugokernel, 2024-01-17:
All PDF files in the hardware/3D section are empty.

shawaj, 2024-01-17:
> All PDF files in the hardware/3D section are empty.

You need to download them and view them in Adobe reader or Adobe acrobat.

They don't work on GitHub viewer as they are 3d pdf files