PiSupply/PiJuice #603: Question: would the point juice work with PoE?

mumia, 2020-12-05:
I ordered this hat to power my pi 4 with PoE. 
https://www.amazon.de/-/en/LoveRPi-Power-Over-Ethernet-Raspberry-Non-Insulated/dp/B07WD7HXSQ/ref=mp_s_a_1_14?dchild=1&keywords=raspberry+pi+4+poe+mütze&qid=1607209178&quartzVehicle=812-409&replacementKeywords=raspberry+pi+poe+mütze&sprefix=raspberry+pi+4+poe&sr=8-14

Would this work with Pi juice? 

I’m aware that the pi juice hides the PoE pins, but my question is, would the GPIO connections power the battery?

shawaj, 2020-12-06:
@mumia it **should** work fine electrically. But we have heard some reports of issues with these kinds of cheap PoE adapters where they don't provide the correct voltage etc etc. So personally I would recommend you use the official Raspberry Pi PoE HAT.

As you say though you will need to find a way to have the GPIO and PoE pins connected and the PiJuice on top.

shawaj, 2020-12-06:
@mumia let us know how you get on with it if possible?

MaxLightstone, 2021-07-28:
@shawaj Is the PiJuice compatible with the standard RP PoE HAT (https://www.raspberrypi.org/products/poe-hat/)?
How would I be able to connect them in series? (I'm a mechanical engineer, and an electrical colleague said that there might be an issue that would require a "power ORing")

tvoverbeek, 2021-07-29:
See https://www.raspberrypi.org/forums/viewtopic.php?t=221479