PiSupply/PiJuice #604: Regarding poweroff

JoshYang1977, 2020-12-06:
I am plugging the power into pi zero and use pijuice to control turn on/off. My question is when the poweroff is activating, both pi zero and pijuice are poweroff. So the power source( portable power supply) is off. Since the pijuice is off and it will not be turned on.

For the consideration of energy consumption, I turn off hdmi/usb/led. However, I turn all on again and it still does not work.

Thank you very much for your answer.

shawaj, 2020-12-06:
@JoshYang1977 I don't understand the issue? Please could you explain more clearly.

JoshYang1977, 2020-12-06:
Thank you very much for your response.

The issue is I use schedule turn on function of pijuice and it is not working. Before schedule turn on, all devices need poweroff. I found out when pi zero is poweroff, the pijuice’ s power is also off (led is off). 

In that regard, it will not turn on on schedule because the power supply I use cannot react to both devices seems totally off.

The function was working before when the led of pijuice was still blinking after the power off command. Right not it is not blinking after poweroff, I think it is somehow totally off.

Hope this helps

shawaj, 2020-12-06:
@JoshYang1977 do you have the power cable plugged into the PiJuice or the Pi Zero?

And what battery are you using?

JoshYang1977, 2020-12-06:
Power cable is plugging into zero and it is a 10000 mah potable power bank. Thank you very much.

shawaj, 2020-12-06:
@JoshYang1977 so you don't have a battery plugged in to the little white 3 pin header on the right hand side of the board labelled J2? Only the power bank plugged into the USB?

JoshYang1977, 2020-12-06:
Yes, only power bank plugs into the pi zero’s usb. Thank you very much.

tvoverbeek, 2020-12-06:
Power banks usually switch off if the load is too low. You need some keep-alive mechanism such as short current pulses every xx minutes.
See issue #210 especially this comment from @mmilann :https://github.com/PiSupply/PiJuice/issues/210#issuecomment-422085393

JoshYang1977, 2020-12-06:
Thank you very much. I just do not know why now the led of pijuice is totally off but before it remained on after poweroff. I do not switch off the lcd.

shawaj, 2020-12-06:
@JoshYang1977 for PiJuice to work you need to have a single cell battery plugged in to J2.

Also note that some power banks, when no charge is being drawn from them, automatically turn themselves off.

JoshYang1977, 2020-12-06:
Thank you all for the responses.

closed 2020-12-06 by JoshYang1977